slic3r/Slic3r #3156: [1.2.10-dev] Slic3r is putting random moves to bottom corner on layer change

lordofhyphens, 2015-12-12:
Printing http://www.thingiverse.com/thing:1071090
The resultant gcode has a move to the lower-left corner of the part on every layer change.

config:
[config-move-to-corner.txt](https://github.com/alexrj/Slic3r/files/60254/config-move-to-corner.txt)

alexrj, 2015-12-19:
Was this with the master or the stable branch?
If it's with master, can you retry now? 667a121ddbbe9732c3831166792128a901c01315 might fix it.

hoh61, 2016-01-24:
I can confirm this effect for revision 1.3.0-dev, downloaded Dec. 8th 2015.

Slicing this part: http://www.thingiverse.com/thing:1133248

[image]

lordofhyphens, 2016-03-13:
Looks like it's fixed for my test model.

closed 2016-03-13 by lordofhyphens