slic3r/Slic3r #3157: Why does nobody get dual extrusion skirts right?

RTimothyEdwards, 2015-12-12:
Everybody seems to agree that skirts are necessary for getting filament to flow properly, putting it down in an otherwise unused spot before jumping to the object when the nozzle is good and primed.  This is one of the secrets to getting the first layer to stick to the bed.

So why doesn't anyone implement skirts properly for dual extrusion?

The idea is that each extruder needs priming separately, and that just like the single extruder, you want to jump from the skirt to the object quickly, while everything is primed and ready to go.  That suggests that the order of printing should go like this:

1. Left extruder prints half of the skirt
2. Left extruder prints first layer of object
3. Right extruder prints the other half of the skirt
4. Right extruder prints first layer of the object
5. Left extruder prints 2nd layer of the object ...

and so forth.  Instead, all the slicers I have tried want to print in this unfortunate order:

1. Left extruder prints half of skirt
2. Right extruder prints half of skirt
3. Left extruder prints first layer of object
4. Right extruder prints first layer of object
5. Left extruder prints 2nd layer of object ...

which is all wrong, because for each extruder, it puts a delay between priming and printing, during which the extruder becomes un-primed, so that when it starts printing the model, nothing comes out for the first few millimeters, and the print's first layer gets messed up, and you're lucky if it all recovers by the third layer or so.

Somewhat less bad, but still questionable, is the way slic3r puts down alternating skirt perimeters with left and right extruders.  Tool changes take time, especially if the filament is drippy and has to be retracted far back, and unnecessary switching between left and right should be avoided at all costs.  Not to mention the fact that if the purpose of the skirt is to prime the nozzle, then why would it need priming several times in a row?

If anybody knows how to fix this, please let me know before I go digging around in the code trying to figure out how to fix it myself.  I am at least happy that slic3r is open-source software, which means that I really can go fix the problem myself if I have to, instead of having to argue back and forth with some obtuse customer support person who just doesn't get it, which is what I've been doing for the past couple of days until I have conceded that, as usual, closed-source software is always the problem, not the solution.

pyrophreek, 2015-12-14:
+1 (million)

I completely agree.  This also results in completely unnecessary toolchanges during the skirt print.  I ended up ditching skirts all together in favor of a wipe routine, but I still never got it to work as well as skirts do for single extruder prints.

RTimothyEdwards, 2015-12-14:
I think that in the short term, this is solvable with a post-processing script, but that's a hack solution and I hope that the issue will be addressed in a more obvious manner in the source code.

RTimothyEdwards, 2015-12-14:
Here's a complicated post-processing script that effectively solves the skirt problem, although it's really just a hack:

[twin_postproc.zip](https://github.com/alexrj/Slic3r/files/61437/twin_postproc.zip)

It removes all the tool changes from the skirt, then divides the skirt into two parts, inserting each part directly in front of the start of model printing, separately for each extruder.

It has another feature that allows for asymmetric tool changes.  It will parse lines of the specific syntax "{[next_extruder]=0} run_this_gcode" or "{[next_extruder]=1} run_that_gcode".  After standard substitutions, slic3r will generate lines with either "{0=0}" or "{1=1}" (matching), or "{1=0}" or "{0=1}" (non-matching).  The post processing script will keep all matching lines and delete all non-matching lines, then remove the match statement so that all that's left is clean g-code.  So you can have the tool change script do one thing when switching from T0 to T1, and do a completely different thing when switching from T1 to T0.