slic3r/Slic3r #3158: Slic3r won't connect to RUMBA board running marlin

mmarkle, 2015-12-14:
When I try to connect to my printer, it shows the port (/dev/ttyACM0), but when I tell it to test (at the correct baudrate, I've tested that), it gives me a connection failed message. Not sure if this is related at all, so I'm going to make another issue about it, but when I click on the Controller tab, it crashes and leaves the message "Segmentation fault (core dumped)" in the terminal.
[config_no_connect.zip](https://github.com/alexrj/Slic3r/files/60749/config_no_connect.zip)
<h6>Stuff I'm supposed to include</h6>
Commit ID: 21364b7cd10fa503bde4b53c53f97c2e76b75203
OS and version: Xfce 4.10 Ubuntu run through crouton on a chromebook.

alexrj, 2015-12-14:
Connection failure on Linux might be a permission issue on the serial port. Can you try running Slic3r as root?

xoan, 2015-12-14:
BTW, there is no relationship between Slic3r and port/baudrate connection with the printer, right?

mmarkle, 2015-12-14:
Ran it as root, now it just crashes when I click test. I'll try a git pull

mmarkle, 2015-12-14:
git pull says it is up to date

mmarkle, 2015-12-14:
Possibly unrelatedly, it also crashes when I load any stl file then click print. It leaves the segmentation fault (core dumped) message.

xoan, 2015-12-15:
OK, I can confirm the same issue using Sanguinololu in /dev/ttyUSB0 @ 115200 baud (I can connect using any other terminal/host application, like screen or Printrun)

Launching slic3r.pl with --debug doesn't show any related information.

alexrj, 2015-12-18:
Thank you, I still have to debug more on Linux.

alexrj, 2015-12-18:
The serial controller in Slic3r is fully multithreaded and implemented in C++, so it's a bit tricker than implementations using higher level languages.

Patola, 2016-07-11:
How can we help debugging this issue? I have a few repraps with gen7, rumba and RAMPS, with repetier firmware, marlin and marlin kimbra. I am still having only coredumps so far when I click on the controller tab and even though it connects when I click on 'test' for the USB/serial connection, it returns 'failed'. With some directions on what to do to debug this I (and others) might be of some help.

alexrj, 2017-03-26:
This is fixed now. I tested on a RUMBA board using Ubuntu 16.04.

closed 2017-03-26 by alexrj