slic3r/Slic3r #3161: Does not slice inner thread

bilderkiste, 2015-12-17:
Slic3r version: 1.2.9
OS: Ubuntu 14.04 LTS
Command line parameters used: no
Slic3r configuration file: all configs on default

When I try to slice the "Caps with grips" part from http://www.thingiverse.com/thing:616895/#files, the inner thread will not sliced.
[image]

mrvn, 2016-03-08:
How thin is the thread in the cap and what is your nozzle size and the widths set for different thread types?

bilderkiste, 2016-04-19:
Hi. My nozzle size is .4 mm. 

I am not sure what you meant with widths set for different thread types. In Slic3r itself I used l the default settings. 

Did you mean with thin (outer diameter - inner diameter) / 2? If yes, this is ca. 2mm.

lordofhyphens, 2016-04-19:
Please provide a config.ini export as requested in the guidelines.

bilderkiste, 2016-04-19:
This .ini files come from a fresh installation. I made no settings. I only loaded the grips.stl file into Slic3r. The problem is still there.

slic3r.ini
`# generated by Slic3r 1.2.9 on Tue Apr 19 10:31:39 2016
autocenter = 1
background_processing = 1
last_version_check = 1461076271
mode = simple
version = 1.2.9
version_check = 1

[recent]
skein_directory = /home/christian/Downloads`

config.ini
`# generated by Slic3r 1.2.9 on Tue Apr 19 10:33:58 2016
avoid_crossing_perimeters = 1
bed_shape = 0x0,200x0,200x200,0x200
bed_temperature = 0
before_layer_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 3
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 50%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_colour = #FFFFFF
filament_diameter = 3
fill_angle = 45
fill_density = 20%
fill_pattern = honeycomb
first_layer_acceleration = 0
first_layer_bed_temperature = 0
first_layer_extrusion_width = 200%
first_layer_height = 0.5
first_layer_speed = 30
first_layer_temperature = 200
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 10
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 80
interface_shells = 0
layer_gcode = 
layer_height = 0.3
max_fan_speed = 100
max_print_speed = 80
max_volumetric_speed = 0
min_fan_speed = 35
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.5
octoprint_apikey = 
octoprint_host = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 60
perimeters = 3
post_process = 
pressure_advance = 0
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 0
retract_length = 2
retract_length_toolchange = 10
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 40
seam_position = aligned
skirt_distance = 6
skirt_height = 1
skirts = 1
slowdown_below_layer_time = 5
small_perimeter_speed = 15
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 20
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\n
support_material = 0
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 0
temperature = 200
thin_walls = 1
threads = 2
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 15
top_solid_layers = 3
travel_speed = 130
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 0
xy_size_compensation = 0
z_offset = 0`