slic3r/Slic3r #3165: Window layout broken

hippich, 2015-12-19:
This is latest 1.3.0-dev master branch build. Something happened to side panel, see below:

[image]

hippich, 2015-12-19:
I am seeing this in the console when launching slic3r.pl, not sure if this is relevant:

```
(slic3r.pl:28191): Gtk-WARNING **: Theme directory  apps/22 of theme unity-webapps has no size field

(slic3r.pl:28191): Gtk-WARNING **: Theme directory  apps/24 of theme unity-webapps has no size field

(slic3r.pl:28191): Gtk-WARNING **: Theme directory  apps/32 of theme unity-webapps has no size field

(slic3r.pl:28191): Gtk-WARNING **: Theme directory  apps/48 of theme unity-webapps has no size field

(slic3r.pl:28191): Gtk-WARNING **: Theme directory  apps/64 of theme unity-webapps has no size field

(slic3r.pl:28191): Gtk-WARNING **: Theme directory  apps/128 of theme unity-webapps has no size field

(slic3r.pl:28191): Gtk-WARNING **: Theme directory  apps/192 of theme unity-webapps has no size field

(slic3r.pl:28191): Gtk-WARNING **: Theme directory   of theme unity-webapps has no size field

(slic3r.pl:28191): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 16

(slic3r.pl:28191): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 16

(slic3r.pl:28191): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 16
```

hippich, 2015-12-19:
I can confirm that removing https://github.com/alexrj/Slic3r/commit/4b30d67e54dcc6ec1d4aff0c09370fde00c1e07f commit fixes issue.

alexrj, 2016-03-13:
Thank you for reporting this, that commit was reverted.

closed 2016-03-13 by alexrj