slic3r/Slic3r #3169: White grid is too bright

a4jp-com, 2015-12-20:
I made this to make the grid dark gray so the text is readable.

https://github.com/a4jp-com/Slic3r/commit/15cd55cf2f450d004f0a2cd957001999b74f7dde

Mikk36, 2015-12-20:
Got comparison images to show?

xoan, 2015-12-20:
What version? In 1.2.9 text is perfectly readable.
[image]

a4jp-com, 2015-12-20:
Version 1.3.0-dev

[image]
[image]

alexrj, 2015-12-20:
Black background is a bug.

a4jp-com, 2015-12-20:
Okay. I though it was a design change.

alexrj, 2016-12-17:
Is this fixed as of current master?

lordofhyphens, 2016-12-22:
I think so, I don't think I've seen the black background for the regular plater for quite some time.

closed 2016-12-22 by lordofhyphens