slic3r/Slic3r #3171: Option to easily set temperature while printing from the controller. Also before printing.

a4jp-com, 2015-12-21:
Slic3r 1.3.0

First I just wanted to say I can now print directly from Slicer and it is amazing. Thank you so much. 

If you add a box under the current temperature values to easily change the temperature values while printing it would be even better.

Temperature:
Nozzle: 212.0 C    Bed: 65 C

Set temperature: 
Nozzle: XXX C    Bed: XXX C     Set    Off

If this could be used anytime including while printing it would be very nice and also useful when changing filament too.

a4jp-com, 2016-01-03:
Is there a way to send G-code?

alexrj, 2017-03-13:
@a4jp-com, we now have both temperature controls and a field for sending G-code manually. Thanks for the input!

closed 2017-03-13 by alexrj