slic3r/Slic3r #3172: Slic3r 1.3.0-dev: Horizontal shells are miscalculated and come out badly on the top and not at all on the bottom.

a4jp-com, 2015-12-21:
Something is very wrong with this print below.

[image]
[image]

I'm using Slic3r 1.3.0-dev on Windows.

Here is the model: http://a4jp.com/3d-files/bugs/snowflake_pendant.stl
And config file: http://a4jp.com/3d-files/bugs/2015-12-21st-06-17-pm-for-github-config.ini

A few questions that are bug related as far as I can see or just my bad settings.

1. How do I fix the bottom solid layer (horizontal shell) that doesn't get printed but shows perfectly in Slic3r?
2. How do I fix the top solid layer that prints out weirdly and doesn't show up properly in Slic3r?
3. What is a good temperature to get the 1st layer to stick better? Should I make to bed a lot hotter at first or the nozzle?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

a4jp-com, 2015-12-22:
I doubled the size of the model and the problem went away so I just need a way to get this small model printing properly now. Any ideas?

ghuser2, 2016-01-02:
On 1.2.9 version I would change Gap Fill or Detect Thin Walls.

curiouspl2, 2016-01-03:
+1 for this, and gap fil is not guilty. detect thin walls is , but disabling it disables also collision checker so prints like this come out ugly. 

see also : 
https://github.com/alexrj/Slic3r/issues/3189 , it looks pretty much same to me.