slic3r/Slic3r #3175: Bridging Regression since 1.2.9

ednisley, 2015-12-22:
Version 1.2.9 produces a bridge aligned perpendicular to the two bars of the central link:

[image]

The current development head (7c8b71012c1) produces a diagonal bridge that doesn't anchor properly:

[image]

The small links also have diagonal bridges:

[image]

The config and (manifold!) STL file:
[Slic3r bridge regression - 7c8b71012c1.zip](https://github.com/alexrj/Slic3r/files/69769/Slic3r.bridge.regression.-.7c8b71012c1.zip)

This looks similar to the problem described in [#2646](https://github.com/alexrj/Slic3r/issues/2646), which was fixed in 1.2.7. In this case, however, *none* of the links have correct bridges.

Running the 64 bit version of Xubuntu 14.04, with the chain mail model generated by the most recent dev version of OpenSCAD.

wolfmanjm, 2015-12-30:
I have also noticed that bridge detection is failing in the current versions. even on simple bridges that are detected in the 1.2.9 release

clough42, 2016-01-01:
I am also having trouble with bridges in 1.2.9.  Attached are two STL files and a config file that generate bridges that terminate or have seams in open space.

These are both real production designs where I had to redesign the part to be able to print it correctly with Slic3r.

[ShroudBridges2.zip](https://github.com/alexrj/Slic3r/files/75851/ShroudBridges2.zip)

Some examples below.  Note the seams in the bridging over open space.

[image]

[image]

[image]

clough42, 2016-01-01:
Okay...the problem I'm seeing is in 1.2.9, so maybe it's something different.

ghuser2, 2016-01-02:
How do you design around a floating bridge seam like that? Add a support bar there?

clough42, 2016-01-02:
I changed the print orientation so bridging want required.  I wanted to print it flat on the bed, using the large base of the post for adhesion, but ended up flipping it so it stands up on the narrow front edges and adding a breakaway plate for adhesion.

curiouspl2, 2016-01-03:
+1 on this one. 
1.2.9 detects bridge direction ok, but there are many 'terminate and start in free space' bridges.

thiagopeixoto16, 2016-01-07:
+1 on this, I had the same problem as @clough42

Vicious-one, 2016-01-22:
https://github.com/alexrj/Slic3r/commit/9fcec107373b6846f3775fe353c7ae29653cf9ac is causing this behavior.

a4jp-com, 2016-01-23:
Thanks for finding that @Vicious-one. I hope it can get fixed soon along with the overfill problem.

alexrj, 2016-03-18:
Thank you, I have fixed this and included a regression test.

@clough42, that's a different thing which is caused by the shape of that object. Slic3r is not very good (yet) at dealing with bridges spanning more supports, like in your case. Basically Slic3r tries to handle each segment of the bridge like a separate bridge, but that approach fails because the supports are too thin. Things will improve on that side, eventually. It's not something I can do quickly.

closed 2016-03-18 by alexrj