slic3r/Slic3r #3176: Version 7c8b71012c1 crashes when selecting Help -> About

ednisley, 2015-12-22:
Running 7c8b71012c1 in 64 bit Xubuntu 14.04:

```bash
/opt/Slic3r-git/slic3r.pl 

(slic3r.pl:6883): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 20

(slic3r.pl:6883): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 20

(slic3r.pl:6883): Gtk-WARNING **: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 20
Error while autoloading 'Wx::wxCLOSE' at /opt/Slic3r-git/lib/Slic3r/GUI/AboutDialog.pm line 69.
```

The first three warnings happen during startup. Without loading an STL file, the last error happens instantly after selecting Help -> About in the GUI menu.

The build-from-source dance didn't report any obvious errors, although there were some warnings while building slic3r-XS:

```bash
cc -I/usr/lib/perl/5.18/CORE -fPIC -xc++ -D_FILE_OFFSET_BITS=64 -Isrc/libslic3r -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/admesh/stlinit.o src/admesh/stlinit.c
src/admesh/stlinit.c: In function ‘void stl_read(stl_file*, int, int)’:
src/admesh/stlinit.c:292:36: warning: ignoring return value of ‘int fscanf(FILE*, const char*, ...)’, declared with attribute warn_unused_result [-Wunused-result]
       fscanf(stl->fp, "endsolid\n");
                                    ^
src/admesh/stlinit.c:293:40: warning: ignoring return value of ‘int fscanf(FILE*, const char*, ...)’, declared with attribute warn_unused_result [-Wunused-result]
       fscanf(stl->fp, "solid%*[^\n]\n");  // name might contain spaces so %*s doesn't work and it also can be empty (just "solid")
```

kefir-, 2015-12-22:
I reported this as issue #2881, and the suggested fix was implemented in pull request #3166. So hopefully it'll be included soon. :)

a4jp-com, 2015-12-23:
Nice work @kefir-

alexrj, 2016-03-13:
Merged @kefir-'s patch.

closed 2016-03-13 by alexrj