slic3r/Slic3r #3179: Refresh/redrawing bug in Slic3r 1.3.0-dev

a4jp-com, 2015-12-24:
[image]

When I start moving things around in Windows 10 everything gets messy.

alexrj, 2016-12-17:
Hm, is this fixed as of current master?

alexrj, 2017-03-05:
@a4jp-com, can you confirm this was fixed as of current master?

lordofhyphens, 2017-03-11:
Looks fixed to me on my Win10 system. @a4jp-com please confirm, I will close this in the meantime (checked on my local Win10 system from a recent build off of bintray).

closed 2017-03-11 by lordofhyphens

a4jp-com, 2017-03-11:
I'll download the new file, compile it and get back to you in the next 30 minutes .

a4jp-com, 2017-03-11:
Everything compiled perfectly and this problem was gone ^^