slic3r/Slic3r #3181: slic3r 1.2.9 craches on slicing model (reports attached)

krokus57, 2015-12-26:
Craches everytime i add model (attached) after several seconds of slicing. I suppose it happens then i set raft = 0 and infill = 0

slic3r-osx-uni-1-2-9-stable.dmg 2015-06-17 10:49 14M
macosx 10.9.5

[LFS_Elephant.STL.zip](https://github.com/alexrj/Slic3r/files/72380/LFS_Elephant.STL.zip)
[config.ini.zip](https://github.com/alexrj/Slic3r/files/72381/config.ini.zip)
[apple.crash.report.txt.zip](https://github.com/alexrj/Slic3r/files/72382/apple.crash.report.txt.zip)

part of report (full version attached too):
Process: slic3r [12905]
Path: /Applications/Slic3r.app/Contents/MacOS/slic3r
Identifier: slic3r
Version: Slic3r 1.2.9.99 (1.2.9.99)
Code Type: X86 (Native)
Parent Process: launchd [163]
Responsible: slic3r [12905]
User ID: 501

Date/Time: 2015-12-26 10:41:27.580 +0300
OS Version: Mac OS X 10.9.5 (13F1507)
Report Version: 11
Anonymous UUID: 81B928D3-22F1-7ADD-8109-1DB400EFD1FC

Sleep/Wake UUID: E3FE298E-2F76-4ABA-A51D-BC3B14EC3533

Crashed Thread: 4 com.apple.appkit-heartbeat

Exception Type: EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x000000001a0cfd40

VM Regions Near 0x1a0cfd40:
MALLOC_SMALL 0000000014000000-0000000016000000 [ 32.0M] rw-/rwx SM=PRV

-->
__TEXT 000000003e000000-000000003e042000 [ 264K] r-x/rwx SM=COW /System/Library/Frameworks/GLUT.framework/Versions/A/GLUT

lordofhyphens, 2016-05-29:
Does not crash in 1.3.0-dev, Linux x86-64.

alexrj, 2017-03-18:
We now have nice stable automated builds for OS X. Get them at http://dl.slic3r.org/dev/

closed 2017-03-18 by alexrj