slic3r/Slic3r #3183: Slic3r 1.2.9 stable 64 bit missing nearly everything under lib/std

thehans, 2015-12-28:
I tried to load an AMF file and got this error:

` no element found at line 1, column 1, byte 0 at /home/peeps/apps/Slic3r-1.2.9-stable64/lib/std/Slic3r/Format/AMF.pm line 19`

Comparing the zips from here:  http://dl.slic3r.org/linux/
I found that the x86 version has a ton of files and subdirectories (97 of theml) under lib/std.  
While the 64bit  version has no files and only 2 subdirs: "Mozilla" and "XML"

maktub8, 2016-02-25:
@thehans Hi bro，have you solved it ?

lordofhyphens, 2016-02-25:
"No element found" is a generic XML error; there was likely something wrong
with your input file (or Slic3r's handling of it)

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2016-03-18:
@thehans, unless you provide a test AMF file to load we can't reproduce this or take any action.

maktub8, 2016-03-25:
Maybe, the AMF file is wrong. I proved it. Thanks everybody.

alexrj, 2017-03-18:
We're working on a new packaging method for Linux, which will fix this anyway.

closed 2017-03-18 by alexrj