slic3r/Slic3r #3184: Feature request: Enable "Bridges fan speed" in advance

thehans, 2015-12-30:
When bridging, I noticed my printer draws a few lines before the fan has a chance to spin up.  It would be nice if the fan was enabled a few seconds before the bridging starts, to give it time to spin up.  Maybe as an adustable parameter.

curiouspl2, 2016-01-03:
+1 to this one. 
also check out : 
https://github.com/alexrj/Slic3r/issues/2221

DrLex0, 2016-12-30:
I would make this more general: there should be a parameter for fan spin-up time, such that the fan can be enabled roughly that much in advance to ensure maximum cooling when it is needed. Actually this kind of stuff should be handled by firmware, but getting good timing estimates for many commands in advance is probably too much asked for the average microcontroller…