slic3r/Slic3r #3185: bottom_solid_layers trumps top_solid_layers in strange ways

kaefert, 2015-12-30:
I'm using the slic3r-linux-x86_64-1-2-9-stable release.
I have these settings:

layer_height = 0.05
fill_density = 5%
bottom_solid_layers = 7
top_solid_layers = 24

The 24 top layers are a more than enough to get from the near nothing 5% infill to a nice surface with the 0.05 layer_height. But my problem is that the 7 bottom_layers trump the top_layers in places (and near them) where there are structures that have less than 31 layers (speculation on my part). For the part that is that thin thats ok since it does not have any 5% infill - but the problem is that thicker parts in their vicinity also don't get the full 24 top layers which ends up as holes in the surface.

markwal, 2015-12-30:
Are you showing the top layer here?  Do you have an stl file to share?  It's hard to tell what is going on from just this picture.  For all we know there are 24 more layers above the parts you circled.

kaefert, 2015-12-30:
no this is not the top layer. This is a layer that shows the problem. The marked area has surface around 15 layers above it just the same as the area surrounding it (it's a building and continues just the way you would expect it to)

I'm sorry I don't own the rights on this stl, so I can't share it.
But I think I can recreate the problem with simply stacking two cubes. I'll check it and report back.