slic3r/Slic3r #3186: Processing Triangulated Mesh

miaoski, 2016-01-01:
Hi.  My slic3r hanged when processing triangulated mesh on this STL: http://www.thingiverse.com/thing:416877

Slic3r 1.2.9
Mac OS X 10.10.5

The config file and STL file are attached.  Thank you!
[bikev3_with_config.zip](https://github.com/alexrj/Slic3r/files/75894/bikev3_with_config.zip)

ghuser2, 2016-01-02:
On larger models mine takes 30 minutes at the Triangulated Mesh stage, but then finishes. Maybe yours is just taking long too.

miaoski, 2016-01-05:
Hi.  I've downloaded Slic3r 1.2.9 running on Windows 7 (x64, 16GB memory, i5 desktop CPU) and it's been doing Triangulated Mesh for more than 16 hours.  Not sure if I should wait longer ... I will give it another 16 hours and come back. :)

a4jp-com, 2016-01-05:
Got the file? I can check it on 1.3.0-dev.

miaoski, 2016-01-05:
Hi a4jp,
Still stuck there.  Kindly try it.  My file is attached in the top most dialog.  Thank you!

munzp, 2016-01-06:
Hi miaoski,

The problem that you are having is the result of a bad STL file and not a Slic3r issue. The file, bikev3.0.stl, has many free edges that require extra elements to repair and thus could not be simply stitched together to make it manifold. The file also containes a number of T junctions where three or more triangles meet at the same edge. The repaired file is included here to allow you to check that your problem has been resolved. 

[bikev30_Repaired.zip](https://github.com/alexrj/Slic3r/files/79285/bikev30_Repaired.zip)

I don't use the usual open source software to fix STL files and can't recommend such software to you, however, a quick search of a relevant reprap forum will point you in the right direction should you wish to understand what is wrong with the original file and help will also be available there that will enable you to repair bad STLs - since you may rest assured that this wont be your last problem file.

Best of luck.

miaoski, 2016-01-06:
Hi munzp,
Thank you very much for the repaired STL file!  It works like a charm!  I will submit the file back to Thingyverse.

closed 2016-01-06 by miaoski