slic3r/Slic3r #3187: Pressure Advance: pressure discharging when doing retracts instead of when the printing speed changes

PvtPrivate, 2016-01-02:
Slic3r 1.2.9 stable
when using pressure advance, the pressure discharges on retractions instead of when the printing speed is changed.

example:

instead of:
from perimeter (20mm/s) to infill (80m/s) pressure advance (0.8mm)
begin printing infill
retract (3mm)
unretract (3mm)
begin printing infill
pressure discharge (0.8mm) from infill (80m/s) to perimeter (20mm/s)
begin printing perimeter

what it's doing:
from perimeter (20mm/s) to infill (80m/s) pressure advance (0.8mm)
begin printing infill
pressure discharge (0.8mm) ??
retract (3mm)
unretract (3mm)
begin printing infill
begin printing perimeter

it basically retracts 3.8mm and unretracts 3mm when the printing the infill

because of this, the infill fails every time it retracts. And the next perimeter will over extrude because it didn't compensate for the pressure difference later.

Pics:
no advance pressure:
[image]

with advance pressure:
[image]

with advance pressure and manually fixed gcode:
[image]

sorry for blurry pictures and weird explanations.

ghuser2, 2016-01-03:
So, pressure advance is not working in 1.2.9 ?

lordofhyphens, 2016-01-03:
Pressure advance has never really worked for me. I think fw retract and/or
volumetric E have something to do with it.
The algorithm is a hack anyway.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

hoh61, 2016-01-03:
I'm impressed from your print results with pressure advance, I never got that far.

Beside that, from my point of view the pressure advance has to be a feature of the printer firmware and not a part of an external software. It cannot react fast enough. 

In fact, there was some years ago an attempt to introduce the pressure advance mechanism in the marlin firmware, it is still there as an unused relict.

lordofhyphens, 2016-01-04:
The algorithm employed in even the sailfish firmware assumed infinite
acceleration on the extruder.

a4jp-com, 2016-01-04:
I'm using 1.3.0-dev and kind of have a similar problem. I think it just might be because I'm using a Bowden tube though. I increased the bridging flow rate and it was nice but small part for example 5x5x5mm cubes get overfilled so something is wrong somewhere.

hoh61, 2016-01-04:
From theoretical point of view a low path system (which the extruding in theory is) can be forced to a rectangular output (extrusion flow) when applying a dirac peak at the beginning and the end.
This is equivalent to a infinite acceleration on the extruder at the beginning and end of extrusion, which should practically be performed with the highest possible feed rate.

I will check the implementation on the sailfish firmware.

PvtPrivate, 2016-01-04:
repetier firmware have advance pressure, but i never got it working for long bowden tubes. The author mentioned its only meant for direct extruders.

hoh61, 2016-01-08:
Just to follow up some former comments:

1) Pressure advance is a feature which is independent on the type of extrusion.

I personally have a bowden type extruder, where pressure build up and relax is taking a long time, therefore pressure advance would be helpful for this type of 3D printers. I checked this with a special test procedure, where I did a variation of the extrusion width. Without pressure advance the printed width and the programmed width shows a large difference and are not correlated.
I was very surprised, that  on a direct extruder printer from a colleague the print result is similar.  Therefore pressure advance will improve for both types of construction.

2) Firmware options for pressure advance
I'm using the Vellemann K8400, where the firmware is based on the Marlin project. There was a small hack for pressure advance, which doesn't work when activated. Therefore I implemented an own pressure advance procedure, which is based on fast (peak) advance at the beginning of an extrusion and fast (peak) relaxation at the end of an extrusion run. It is just working quite well, and I could achieve a better linearity between theoretical line width and extruded line width. My very first tests are quite good, I will check some further prints.

3) Pressure advance options in slicer
My first attempts to use the implemented functionality within SLic3r failed, nevertheless, it was shown, that it is capable tom improve the print results.
My wish would be to implement a material parameter "Pressure advance", where for each material a specific command can be stored, which will control the firmware setting for pressure advance.

lordofhyphens, 2016-01-08:
Except that pressure advance isn't common to firmware. Because it is a hard
problem.

hoh61, 2016-01-08:
... until someone comes and presents an easy solution.