slic3r/Slic3r #3189: [1.2.9-stable] weird overextrusions in thin walls area

curiouspl2, 2016-01-03:
Hello. try slicing "top" part of : 
http://www.thingiverse.com/thing:1077430/#files
with extrusion width settings : 
external perimeters = 0.7
infill = 0.7 

you should get something like : 
![thinwall_problem](https://cloud.githubusercontent.com/assets/6697487/12077825/f3ad7838-b1c4-11e5-8f48-ea43291c3127.jpg

width of overextruded mess seems to be proportional to extrusion width of infill. 

this is especially annoying when using scanned/sculpture objects where you cannot easily find all thin walls/correct them. note overextrusion varies from model to model due to some mysterious bug - sometimes it's even larger, making lot of mess.

lordofhyphens, 2016-03-20:
Found a similar problem, seems related to the layer height.

[image]
[overinfill.zip](https://github.com/alexrj/Slic3r/files/181103/overinfill.zip)

Setting the layer height to 0.1 instead of 0.05 causes the outer ring area to render as below:
[image]

alexrj, 2016-03-26:
@curiouspl2, I tested your file with current master (1.3.0-dev) and the following command line options:
`--external-perimeter-extrusion-width 0.7 --infill-extrusion-width 0.7`

I get a clean result:
[image]

I also tested with current stable (1.2.10-dev) and I get this:
[image]

Do you have a chance to test with one of these git branches and confirm this is fixed?

@lordofhyphens, please let me know if yours is fixed as well now.

lordofhyphens, 2016-03-26:
The thin gapfill has gone away (to be replaced by nothing), but the overextrusion screenshot was not gapfill; it was regular infill and that is still present in 9576973b578f0c39f8cbd73c225187ba65b6cef4.

alexrj, 2016-03-27:
Thank you for testing @lordofhyphens

lordofhyphens, 2016-03-27:
@alexrj it appears that the massive overfill on the infill that I noticed
occurs when thin wall detection is on, and it's definitely not gapfill
(gapfill speed is set to 0)

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-07-12:
#3196 may be related to this, at least regarding the solid infill width being calculated incorrectly.

alexrj, 2016-12-21:
@lordofhyphens, I tested your overinfill.zip file with current master and I confirm the issue is fixed:

[image]
[image]

When enabling gap fill:
[image]

closed 2016-12-21 by alexrj

lordofhyphens, 2016-12-21:
Excellent. Any idea what this was (other than perhaps the issue with that function not being idempotent)?

alexrj, 2016-12-21:
I think that was the issue, because those extrusions were huge and that issue caused width to grow at each cycle. Other issues related to extra gap fill had other overlaps but not such a huge width...