slic3r/Slic3r #3191: Slic3r no longer runs

grtbaldini, 2016-01-04:
Working with it for the past week the latest version kept crashing while saving a new config file and trying to setup a new file to print. I'm using the 64 bit version on a win 7 64 bit machine. Today after another virtual memory shut down, it will no longer run. I rebooted machine, got the 32 bit version, tried older versions, the exe file will not run. Any idea as to what I can try next?

lordofhyphens, 2016-01-04:
Probably something to do with your local system. Don't know what it could
be.

grtbaldini, 2016-01-04:
It seems that windows has blocked it maybe? Pronterface runs but can't call up slic3r inside it either. I checked my firewall and made sure it was trusted but the system suffered a few virtual memory errors when slic3r was running. Blue screen, restarted.

grtbaldini, 2016-01-06:
All I know is my whole printer setup didn't like the pc it was connected to. I recently built a htpc for the house and for whatever reason, the ardiuno mega also didn't like something on that pc because that went into an error on its own. My printer was singing a different tune when it printed, or I should when it wanted to print. So I brought in a different pc and it was like magic. Things are humming like they used to and Slic3r gets up to the plate when called. I'm happy :-) I'll have to investigate the htpc and see whats going on there.

closed 2016-01-06 by grtbaldini