slic3r/Slic3r #3192: Variable thickness on slanted surfaces

10-p, 2016-01-04:
Hello,

We're having some issues with slanted surfaces. Slic3r creates some bedding on some slanted surfaces but not on others (the typical bedding used in ceiling perimeters).  

We have set up all the information we could in this [dropbox link](https://www.dropbox.com/sh/akh2qpzk65yrmb1/AACYcbcJfSnooCxSo2sazPfMa?dl=0).

As you can see in the photos, there is one triangle with no such bedding, making it much more translucent, displaying the infill.

Is it possible to extend such bedding to all of the slanted surfaces?

Right now, we can increase the wall thickness if we want to keep such surfaces thick as well, but that would waste material as the walls are fine as is.

Thanks

Paul--W, 2016-01-08:
P10,
I note that you are using Extra perimeters as needed, and 4 perimeters.

What I've seen from some users on reprap.org is when the slope gets closer to horizontal that outside sloped wall gets thinner.  

Am I correct in thinking you want that sloped outside wall thickness to stay at the same thickness as the vertical walls (width of the 4 perimeters)?   I think it should too, especially if you have "Extra perimeters if needed" on (right now "Extra Perimeters..." only adds perimeters to avoid gaps - according to the tooltip).

10-p, 2016-01-13:
Here is a picture explaining exactly what we want to do.
[image]

As you can see, the perimeter thickness of the slope is exactly 4 which is consistent with configuration.
However, in the lower part of the image, the slope only has this perimeter where in the upper part of the picture, the slope has this additional bedding for the upper layers.

We want to extend this bedding to the upper and left part of the shape (where the slope is), not just stopping midway of the left wall.
[image]

Is it possible to extend said thickness  without touching the number of perimeters?