slic3r/Slic3r #3195: Bad bridging handling

dexy31337, 2016-01-05:
Hi, I've encountered weird slic3r logic, printing lego-like model, having big unsupported (bridged) flat surface, here how it looks like cut with section in the middle:
[image]
Sliced, model appears to have middle section as a separate bridge with edges in the air, which obviously does not get enough adhesion:
[image]
For example, this is how simplify3d handles same model:
[image]
As you can see, all bridge edges are residing on perimeters. This is tested on 1.2.4,1.2.7,1.2.9 with same results. 
Model and config:
[slic3r_bridges.zip](https://github.com/alexrj/Slic3r/files/79046/slic3r_bridges.zip)

wolfmanjm, 2016-01-05:
maybe related to #3175

lordofhyphens, 2016-08-24:
Looks like what Slic3r is doing here is detecting the large flat area and generating perimeters accordingly. 

What S3D seems to be doing is extending those perimeters "up" a layer and then generating bridges. 

An alternative solution I can see is if you have a surface like that, is to ignore the interior area and generate your bridge from the outer perimeter. Basically, use the perimeters from the previous layer as support struts.

lordofhyphens, 2016-08-24:
@dexy31337 could you give this a try?

https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches/1.3.0-bridge-ignore-lower-layer-168

I disabled one of the checks for the previous layer (just a quick+dirty hack). This is the result on your test piece.

[image]

bubnikv, 2016-10-10:
This has been fixed in the Prusa3D build, with following check-ins:
https://github.com/prusa3d/Slic3r/commit/3a81e6bee420196698e3f1409a8a7318dd35965b
https://github.com/prusa3d/Slic3r/commit/d2635ab79944917330dcf50ee7e3d21f7f576e4e

The trick is to merge overlapping bridging regions before the bridging direction is calculated.

he29-net, 2017-02-01:
I can confirm this issue in v1.2.9 on another Lego-like brick model:
[image]
[image]
[image]

I'll try to test the current development version and / or the fix mentioned by @bubnikv as it seems to be exactly the thing that would help.

Also, bugs #3108, #2714, #2506 and #2262 all seem to refer to the same issue. #3578 seems to be also related to this issue (I encountered that one as well, few months ago on a similar part).

he29-net, 2017-02-02:
Well, I tested it with current head (that would be d41e3c1ee4f14f... I guess?) and the result is a bit different, but still bad. 
[image]

As for the prusa3d fork, I had no luck getting it to run; the commit prusa3d@d2635ab fails at some C code errors and more recent versions from January appear to be built OK, but fail with "Can't locate Slic3r/XS.pm in @INC" when I try to run it.

bubnikv, 2017-02-02:
You can compile the Prusa3D fork with the environment SLIC3R_STATIC=1 set.
If not, please provide a bug report.
You may also try the latest release. The bug has been fixed there already.

The fix is in BridgeDetector.{cpp,h} and in
LayerRegion.cpp, function LayerRegion::process_external_surfaces(const
Layer* lower_layer)

The change is extensive though.

he29-net, 2017-02-02:
OK, I got the prusa3D fork to work (the most recent version) and it indeed manages to bridge this complex model perfectly:
[image]

Great work @bubnikv, thanks! As for the build problems, it turns out I was just using "wrong" build command. I thought they would be the same for Slic3r and the fork and only after seeing prusa3d#103 I realized that's not the case. Adding --sudo parameter solved the issue.

lordofhyphens, 2017-02-02:
That means that you have probably polluted your distro's CPAN/Perl setup,
by the way.

lordofhyphens, 2017-04-03:
@dexy31337 @he29-net 
Please retest with current master, there's been some changes to bridging logic.

he29-net, 2017-04-13:
Current master (5c2a05cd) does not seem to be handling this issue any better:
[image]

trzeci, 2017-04-16:
Another example of weird bridge edges on current master 1857bf63910a7ade053390157fbd1bb9533629a0

[image]

trzeci, 2017-04-16:
More simple example:

[image]

I think bridges should check if there is possibility to reach another supporting edge if extended.

trzeci, 2017-04-16:
Strange thing, this example does not cause problems:
[image]

trzeci, 2017-04-16:
Any rotation of supporting edges spoils:
[image]

Sorry for many comments, but I realized new ideas to target a problem.

trzeci, 2017-04-18:
weird: master with SLIC3R_STATIC=1 does not build .. missing Xs.so

trzeci, 2017-04-18:
Where could I find code cutting layer to regions ?
There is logic which extends region beyond supporting regions of lower layer.
It should stop on support if it is internal region.
Problem arises when regions are getting different infill angle, the regions do not merge.

trzeci, 2017-04-18:
OK, I know how it works. There is setting in libslic3r.h EXTERNAL_INFILL_MARGIN set to 3mm.
Every bridging region is extended over supporting layer by 3mm, even if the layer is shorter.
After setting EXTERNAL_INFILL_MARGIN to .3mm effect is a bit better:

[image]

lordofhyphens, 2017-04-18:
How does the anchoring work with bridging like that?

I suspect the "true" solution here is to grow the regions out more aggressively, but idk for sure.

trzeci, 2017-04-18:
Of course you are right. Ideal solution would be to extend bridging region by half of a support.
I'm learning the code now.

lordofhyphens, 2017-04-18:
@trzeci Glad to hear it, I will look over any PRs you produce. I just haven't had occasion to look at this area myself yet.

lordofhyphens, 2017-04-18:
This has been claimed to have been fixed in the Prusa3d branch as of:
prusa3d@3a81e6b
prusa3d@d2635ab

Unfortunately they can't be cherry-picked so porting needs to be done by hand.

From my reading of the work it looks like he did what I thought should be done, and aggressively bridge to the outer perimeters (instead of to the nearest perimeter in the lower layer). 

Alternatively, going the other way (bridging to the center line of the lower perimeter instead of a fixed distance) would probably work as well. Similar to your last picture, it would still produce shorter overall bridges but it wouldn't then try to end them in the air.

trzeci, 2017-04-18:
That's the point, dividing bridge at the center creates shorter bridges.

lordofhyphens, 2017-05-04:
Moved from #3924
[image]

[project and config.zip](https://github.com/alexrj/Slic3r/files/970643/project.and.config.zip)