slic3r/Slic3r #3196: [1.2.9-stable] Heavy over-extrusion with test case

felicitus, 2016-01-06:
Hi,

slic3r 1.2.9 stable heavily over-extrudes in small areas. The file I'm slicing is [WADE-SMALL-GEAR.stl](https://github.com/felicitus/Prusai3_EINSTEIN_Reworked/blob/master/Pieces%20en%20STL/EXTRUDEUR/WADE-SMALL-GEAR/WADE-SMALL-GEAR.stl) and my config is here: [config.ini](https://github.com/felicitus/configs/blob/master/slic3r/config.ini)

The result looks like this:
[image]

It's a bit hard to see, but maybe it gives you an idea about the over-extrusion. I stopped the print when I noticed the issue. Over-extrusion on that particular part has been happening earlier and always happened when printing the teeth section of the gear.

lordofhyphens, 2016-06-20:
Need a layer screenshot from slic3r's own preview to have a prayer of debugging this any further.
I suspect it's gapfill though (which has been fixed). 
Please try disabling gapfill (set the gap fill speed to 0) and see if it still fails.

lordofhyphens, 2016-07-12:
@felicitus  It's not gapfill, looks like the autodetect for the width of the solid infill is getting set too wide. I've seen this with a few other models, usually involving a narrow neck. A usable workaround is to manually set the solid infill width to something smaller (in this case, 0.4 is sufficient to not cause overextrusion).

Slic3r's preview is smart enough to show you the problem, so pay attention to it and tweak the extrusion widths if Slic3r doesn't give you optimal results.

lordofhyphens, 2016-07-12:
Wherever it is, it's attached to whatever is being used to calculate the top solid flow. I haven't been able to find out where this is, but I think if I find the autocalc for the default generation I'll have my answer.

kyrreaa, 2016-09-09:
I found a similar issue with gap fill.
A test I did with 4 perimiters set in config, but circular wall was only thick enough for 3 inner and 3 outer, leaving a very narrow gap. The gap was filled by a flow that is twice that of the normal perimiters!
(For every segment in the circle it used 0.05mm 1.75 filament for perimiters, but the very narrow gap was filled by 0.1mm extrude per segment. Result was the plastic did not fit the narrow channel and "bubbled up" behind the nozzle path causing havoc on next layer. Print quality really suffered.
Tested with 1.1.7 which simply did not fill this gap. Then adjusted perimiter to get it to not need the fill for a good print.

Maybe the same bug can calculate wrong flow on the normal infill?
I have also seen other prints fail due to excessive over extrusion in 1.2.9, up to 10x flow.  (At a normal printspeed where the lengths usually got a flow that equated nicely to the cross sectional volume, suddently it would put down a flow of redicolous proportions.)

Will try and create a testcase.

bubnikv, 2016-09-09:
There is a routine inside the rect infill, which  modifies the flow to be
an integer division of the bounding box width. This makes the wildest
changes in the flow for narrow regions.

bubnikv, 2016-09-23:
You could try a following patch:

@@ -233,6 +233,8 @@ sub make_fill {
             $f->spacing($flow->spacing);
         }

+        my $old_spacing = $f->spacing;
+
         $f->layer_id($layerm->layer->id);
         $f->z($layerm->layer->print_z);
         $f->angle(deg2rad($layerm->region->config->fill_angle));
@@ -243,6 +245,7 @@ sub make_fill {
             $_,
             density         => $density/100,
             layer_height    => $h,
+            dont_adjust     => 1,
         ), @{ $surface->offset(-scale($f->spacing)/2) };

         next unless @polylines;
@@ -254,6 +257,9 @@ sub make_fill {
             # so we can safely ignore the slight variation that might have
             # been applied to $f->flow_spacing
         } else {
+            if (abs($old_spacing - $f->spacing) > 0.3 * $old_spacing) {
+                print "Infill: Extreme spacing adjustment, from: ",
$old_spacing, " to: ", $f->spacing, "\n";
+            }
             $flow = Slic3r::Flow->new_from_spacing(
                 spacing         => $f->spacing,
                 nozzle_diameter => $flow->nozzle_diameter,

This effectively disables adjustment of the infill line width. My opinion
is, the line width adjustment as it is implemented, shall be disabled until
it is fixed.

Vojtech

lordofhyphens, 2017-06-04:
@felicitus can you test with current 1.3.0-dev (http://dl.slic3r.org/dev)? 

It looks like it's been fixed, at least according to the preview. Used your config and model.
[image]

If you are still seeing problems I will reopen.

closed 2017-06-04 by lordofhyphens