slic3r/Slic3r #3197: [1.2.9 stable] Support Material to close

samu87, 2016-01-07:
Hi all, i use slic3r after i had try a lot of slicer before. It's amazing but the Support Material generated are to much close to vertical wall in x and y and it's impossible to clean the printed part from the support. There is a way to put the support far to the vertical wall? A parameter o work around?!

Many thanks!

lorenzo-franco, 2016-01-07:
I have the same problem. Thanks Samu for this issue!

anokiernan, 2016-01-10:
I have also recently been struggling with this issue, it would be very helpful to be able to adjust the XY plane support clearance with a parameter.

lonelymyp, 2016-01-10:
https://github.com/alexrj/Slic3r/issues/2916

CodeChief, 2016-01-10:
Mee too. We have contact Z distance but no "contact X/Y distance". Please add it!

lordofhyphens, 2016-02-15:
So long as I can figure out how to cleanly get a config option propagated to the logic, line 531 in lib/Slic3r/Print/SupportMaterial.pm seems to govern the XY separation. 

Specifically, it's for the interface/base layers, and then the rest of the support is drawn down from that. Looks like you could add an extra constant there (would be in mm so far as I can tell) to reduce things. 

Default is the extrusion width, by the way. So you could increase the extrusion width to get better XY gaps.

lordofhyphens, 2016-03-17:
@samu87 can I close this as it's already covered in #2916 (and related to #1539)?

samu87, 2016-03-17:
Yes many thanks i follow the other 2 post!

lordofhyphens, 2016-03-17:
Closing because duplicate issue.

closed 2016-03-17 by lordofhyphens