slic3r/Slic3r #3198: 1.2.9 x64 in Windows and Linux Sequential Printing NON-FUNCTIONAL

drifter57, 2016-01-07:
I am unable to get sequential printing to function in either the Windows x64 or Linux versions of 1.2.9 Slic3r.  I have set the individual objects switch in the output options.  I have made sure there are no collisions on the plater.  No matter how I configure the plater, Slic3r always generates the g-code as if there is only one part.  I have loaded the .stl file twice so that Slic3r sees two objects on the bed.  I have even tried renaming my .stl file so there are two different files loaded.  I also tried using the .amf file format to see if that would help.  All to no avail.  This feature has worked for me in the past, but I don't remember which code revision.  Have I forgotten some step in the setup, or is there really a problem with the code?  Thanks for looking into this.

drifter57, 2016-01-07:
Here is my original .amf file and my configuration file.
[NewArchive.zip](https://github.com/alexrj/Slic3r/files/81853/NewArchive.zip)

lordofhyphens, 2016-01-21:
Could you try it again with a couple other -dev version builds? I think it'd be useful to know where about this broke.

drifter57, 2016-02-25:
The last version I used that worked was 1.2.6, but when I go back and try to make it work again I can't get it to work in that version either.

Here is the last .gcode file I created that worked.

[Big Vent Cap 4x9 Spread.zip](https://github.com/alexrj/Slic3r/files/146042/Big.Vent.Cap.4x9.Spread.zip)

drifter57, 2016-03-08:
The other day I was able to get sequential printing to work for one pair of objects in Slicer 1.2.9, but it only worked ONCE.  When I tried to use it again for the next set of objects it went back to generating code as if there was only one object on the bed.  This problem appears to be intermittent, and may have something to do with the order options are being set in the software.  I have not yet discovered the set of circumstances which allow it to work and which disable the feature.  Selecting and de-selecting the sequential box appears to have no effect on the output... sometimes.  Checking the setup .ini file shows the feature enabled and disabled as set by the checkbox.  Is there a requirement for location of the command in the .ini file that is affecting this feature?

lordofhyphens, 2016-03-15:
So far as I'm aware, the position of the lines in the .ini aren't sensitive to anything. At least, they shouldn't be. I guess you could fiddle with it until it works and then permute the lines in the config and see if it breaks?

drifter57, 2016-03-29:
Here is a 4 piece print where I tried to have Slic3r 1.2.9 print sequentially.  Again, it failed.  I have included the .stl file, the .gcode file generated and the setup .ini file in the zip.  Please let me know if there is anything else required to help you figure this out.  Am I not inserting parts onto the bed correctly?  Is there some other setting I am missing?

[Dovetail.zip](https://github.com/alexrj/Slic3r/files/194467/Dovetail.zip)

drifter57, 2016-04-19:
Is the information I left 20 days ago useful at all?

lordofhyphens, 2016-04-19:
I have not had time to look at it myself, I will try to reproduce your issue this weekend.

drifter57, 2016-04-29:
Could it possibly be an issue with a command problem on an AMD FX processor in my laptop?  Don't know as I can trust AMD to make the perfect Intel clone...

universam1, 2016-07-15:
Same problem here but on a Intel i5. So not AMD to blame...

drifter57, 2017-01-13:
Just checking to see if there has been any progress on fixing this issue?  I would really like to be able to use the sequential print feature, as it is extremely useful for printing a quantity of small parts.

lordofhyphens, 2017-01-13:
I haven't looked that closely at it, been very busy. Does 1.3.0-dev show you more than one object in the 3D preview?

lordofhyphens, 2017-01-13:
Okay, I just created a test case with four objects and counted the number of Z moves to Z=5.95 using your config from NewArchive.zip. 

There were four moves, which to me says there are 4 objects there. The Preview says there is 4 objects with gcode being generated for it. Same with the first layer (0.23). Same as in your provided Dovetail.gcode. Pronterface's preview also shows four objects, and that wouldn't lie about anything (each object is rendered from gcode). 

Here's the output gcode from 1.3.0-dev. It's not going to match 1.2.9's output exactly due to changes in rectilinear infill.
[output_gcode.1.3.0.zip](https://github.com/alexrj/Slic3r/files/703571/output_gcode.1.3.0.zip)

Here's the output from yagv (yet another gcode viewer) for your gcode (Dovetail).
[image]

How did you verify that only one object had gcode generated for it?

lordofhyphens, 2017-01-13:
The only thing I haven't done is run a print myself.

lordofhyphens, 2017-01-15:
Closing because of non-reproduction. Will reopen if @drifter57 provides additional evidence (perhaps video of the print?) of it failing in current master.

closed 2017-01-15 by lordofhyphens