slic3r/Slic3r #3199: Squish - first layer height feature request

RogerRepRap, 2016-01-08:
When you print with a first layer height of say .3 the initial Z move is to .3 and slicer slices the model as though .3 had been removed from the bottom. This does not take into account the squish required to get the filament bonded to the bed. If the actual height of the first layer is say .25 to accommodate .05 squish my model will come out .05 short on the Z. Depending on material and bed surface may be way more than .05 mm.

There needs to be 2 fields for first layer. Requested and actual or maybe requested and squish distance. From these numbers Slic3r could adjust itself to have the Z height come out right.

Not a big issue but I had some problems with some precision assembly clearances. Its amazing that RepRap printers can be precise enough that this matters.

Roger

a4jp-com, 2016-01-10:
Printer settings>  Horizontal shells > Bottom: Add an extra layer here
Printer settings> Layer height> first layer: Height you want here
Printer settings> General> Z offset: Squish

or

Print settings> Support material> Raft layers

lordofhyphens, 2016-01-21:
Personally, this is down to machine setup, as it is going to be wildly different from printer to printer and even filament to filament. 

I myself do my best to make sure that layer 1 is what I told it to be with as little squish as I can (if I am printing a 0.3mm first layer, I want my first layer thickness to measure to 0.3mm) and use a printing substrate like Printbite or PEI or even beer spritzed onto the bed and heated bed to make sure it doesn't lift off.

mrvn, 2016-03-08:
A first layer extrusion multiplier would be nice to give some extra squish.

vlast3k, 2016-03-08:
what i do is
G28
G92 X0 Z0 Y0 ; to reset from previous prints
G92 Z0.2

Which will move the head 0.2mm deeper

alexrj, 2016-03-18:
@RogerRepRap, so you'd like to print at Z = 0.3mm with more plastic than you'd need for a 0.3mm layer. However, since the bottom layer is solid (= 100% filled), I can't see how you could physically stuff more volume than what's possible. Plastic would overflow or create blobs. Any addition to Slic3r in that direction would be a nasty hack. Adhesion problems should be addressed with print bed surface material, bed leveling, higher temperature, higher bed temperature and thicker First Layer Extrusion Width.

closed 2016-03-18 by alexrj

poikilos, 2019-12-17:
@RogerRepRap @mrvn @alranel I had some separation of lines and found this issue, which lead me to "Z Offset," which is working fine for me. I didn't have to change the extrusion multiplier from 1. I'm not sure when "Z Offset" was implemented, If I calibrate my bed so the nozzle is 0.1mm from the bed (I use a feeler gauge now, but the "average" piece of paper is also 0.1mm), I set Z Offset to -0.1 my first layer is now solid. What I would call "squish" is any "Z Offset" value other than how you calibrated your bed, which is -0.1 for me (squish would be something like -0.125). My bed is warped, so the corners of my bed are 0.15 from the nozzle, but the problem doesn't seem to cause lines of filament to not bond until about 70mm from the edge of the bed, and only one separation exists every few lines (tested using a 1-layer print .32 high). I could set the Z Offset to -0.125 to get enough "squish" to deal with the warped bed. Mesh bed leveling or a better bed would solve that as well.