slic3r/Slic3r #3200: Complete individual Object temperature problem

IvanBortolin, 2016-01-08:
Hi Ale, 
I think there is a bug in temperature control in mode "complete single object". If in your filament settings you have two different temperature for first layer and another layer, when finish first object and start second object, the printer set first layer temperature, but at second layer don't reset correct temperature.

a4jp-com, 2016-01-09:
That sounds bad. I hope 1.3.0-dev doesn't have this problem. What version are you using?

IvanBortolin, 2016-01-09:
Hi, sorry Ale. By mistake I used version 1.2.6 dev. Now I try whith 1.2.9 and it works well! Thank.

a4jp-com, 2016-01-09:
Nice. Problem solved ^^

a4jp-com, 2016-01-10:
Can you close this thread if the problem is solved?

IvanBortolin, 2016-01-13:
Thank Ale!

closed 2016-01-13 by IvanBortolin