slic3r/Slic3r #3201: BadValue (integer parameter out of range for operation)

BeebBenjamin, 2016-01-12:
Hi I am using Ubuntu 15.04 64bit and I clone the repository using the instructions in the readme.  When I try to load up Slic3r I get the following output:

The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 793 error_code 2 request_code 154 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

Any idea what is causing this?

solucaocriativa, 2016-10-29:
Hi, I'm facing the same issue, on Ubuntu 16 64bit:

The program 'slic3r.pl' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 582 error_code 2 request_code 154 minor_code 24)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

lordofhyphens, 2016-10-29:
These are usually library/driver issues with OpenGL, if memory serves. Last
time I remember seeing something similar is when I was trying to a
forwarded X session.