slic3r/Slic3r #3202: [1.3.0-dev] Move Controller tab

sorki, 2016-01-12:
Placement of this tab is pretty bad as I was used to automatically click on the next tab after the plater but now it causes whole slic3r to crash. It would be better to move Controller tab out of the way (e.g. after Printer Settings) or to the bottom tab group not to break workflow we are all used to.

alexrj, 2016-03-18:
The solution is to fix the crash, rather than moving it away... The GUI in the master branch is under redesign, so this is not the final layout. Things do change, however, and habits can be changed as well - especially when it's just matter of clicking a few pixels away ;-)

closed 2016-03-18 by alexrj

sorki, 2016-03-20:
And it's just a few lines away from driving people crazy by putting a tab that causes crashes in place of probably most used tab in whole slic3r.