slic3r/Slic3r #3203: Can't connect to octoprint over SSL

taxilian, 2016-01-13:
I have my octoprint server exposed via an SSL port so I can access it remotely, but slic3r can't use it.  Is this something that can be trivially fixed? Would be cool =]

Awesome tool, keep up the good work.

lordofhyphens, 2016-01-13:
The connection code, IIRC, uses the Perl cURL library.  Should be fixable
in code w/o much trouble.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

taxilian, 2016-01-13:
If you can give me an idea of where to start looking I may be able to fix it myself?

lordofhyphens, 2016-01-13:
It'll have to wait until I can get in front of the codebase.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

taxilian, 2016-01-13:
Meh, I'm currently sitting in a datacenter re-installing servers; as much as I'd like to, I probably shouldn't be playing with my 3d printer right now anyway =]  No rush, just wanted to let you know I'm willing to help as I can to solve the problem.

Richard

lordofhyphens, 2017-03-28:
Moving to 1.3.1, I'd like to do this in xsgui instead of Perl.

keithellis74, 2018-01-05:
Probably related to this.  I use OctoPi as installed with the [dietpi](http://dietpi.com/) distro,  By default this opens up the OctoPi instance on port 5000.  Slic3r does not have an option to change the port.  even if the IP address 192.168.xxx.xxx:5000 is used, it appears just to be checking port 80.  The following error message is received.
﻿
[image]
﻿
Can a port option be added please.

Stiixxy, 2019-01-28:
Just to let everybody know, i started working on this yesterday and got proof-of-concept fix done.
I'm currently in the proces of improving it and making it configurable.  Since @lordofhyphens posted this 

> 
> 
> Moving to 1.3.1, I'd like to do this in xsgui instead of Perl.

Should I still make a pull request with the fix?

lordofhyphens, 2019-01-29:
@Stiixxy did you have to add a new Perl module for a dependency?

Stiixxy, 2019-01-29:
@lordofhyphens yes i beleive it required the LWP::Protocol::https module. I'm gonna recheck it when i'm working on it again. I got a https fix and http credentials for basic authentication because that's how my octoprint is setup.