slic3r/Slic3r #3205: Feature Request: Ability to Determine Print Order of Sequentially Printed Objects

drifter57, 2016-01-15:
The ability to print sequential objects in a Z or perhaps W pattern.  

My printer has the majority of its mass on the right hand side of the print head, mainly because of the cooling fan on a Greg's Wade type extruder.  If it were possible to print parts down the left side of the Y axis, move right, print up the Y axis, move right, print down the Y axis, etc., then I would need less clearance for each part and could put more on the platen for each run.

The ability to print parts in ANY order required would also be a nice feature.

Thank you for your consideration.

mrvn, 2016-03-08:
You can always place the parts manually so it prints in that order.

But for a more general solution it would be cool if one could upload an STL file for the hot-end shape, just like one can upload an STL file for the print volume. I believe non-symetric hot-ends are pretty common due to a cooling fan on one side. Or dual extruders. radius+height is a rather simplified way to specify the hot-end dimensions.

Even if the hot-end is symetric many printers have a gantry. They can print multiple objects of any height going in Y from back to front (or front to back). But put 2 objects in X next to each other and the gantry limits the height.

drifter57, 2016-03-30:
The gantry issue would be a reason to allow the reverse Z pattern printing from the left front corner across the X axis, moving back and repeating till you get to the back right corner.

I have tried placing the parts manually, but the slicer apparently picks them at random when setting their sequential priority.  I have to run the slicer, then determine the order it has picked. Rearrange the parts in the order I want and slice again to make sure it has not re-randomized.  Arrange again until the slicer finally processes them in the order I want them to print.  Sometimes takes several iterations.  

If there were also an align command that would just line up parts on X and Y without rearranging them to their initial placement positions that would be a magnificent improvement.

Lately however, I have not been able to get Slic3r 1.2.9 to generate gcode for sequential printing at all.  It generates everything as if it is one part no matter what the settings.

mrvn, 2016-03-31:
On Tue, Mar 29, 2016 at 05:45:15PM -0700, drifter57 wrote:
> The gantry issue would be a reason to allow the reverse Z pattern printing from the left front corner across the X axis, moving back and repeating till you get to the back right corner.
>
> I have [...]

I'm not sure what you are doing but so far slic3r has always printed
my objects in the order listed on the right in the GUI, in the order
you loaded them. If you split an object you get a random order but you
can split an object, delete all but the first part, load the object a
second time, split again, delete everything but the second part and so
on.

MfG
	Goswin

nenadalm, 2016-05-13:
:+1: would be nice to be able to specify order (or upload the stl with specifying gantry height)

Problem with random order also appears to be when you increase number of copies (so there is 1 item on the right in the gui).

bubnikv, 2016-05-14:
Currently Slic3r sorts the objects by their maximum height. The shortest object gets printed first, the tallest last. If the objects are of the same height, then I suppose the order will be given by the internal order of the objects in Slic3r, in which they were loaded.

I suppose you can enforce a print order of the set of equal STLs by adding a very thin needle at the top of your STLs. The objects will then be printed in the order of their increased needle height. Not the most convenient solution, but at least you have some.

Vojtech

PR #4153 (2017-10-17): Added option to manually reorder sequentially printed objects via plater
Sequential prints can now be reordered manually by assigning a 'sequential print priority' value (default = 0) via the object settings dialog.

Sequentially printed objects are ordered first by sequential-print-priority, and then by object height -- this means Slic3r defaults to current behavior if sequential-print-priority is not explicitly set by user. The ordering is currently still implemented in Perl, with plans to replace with C++ logic when I get around to implementing object merging in the near future.

Print::val [... cut]
closed 2017-10-17 by lordofhyphens