slic3r/Slic3r #3206: 1.3.0-dev: support material layer variation

hoh61, 2016-01-16:
I was trying to generate a slicer file for a structure using PVA as support material. Therefore I selected the option for an initial support layer and support material touching the structure.

I was very surprised about the final layer generation at a detailed look at the layer preview. The extrusion layers are listed for the support material left, for the stucture extrusion on the right side:

Support   Structure
0,2                              ; first layer support material
0,4        0,4
0,6        0,6
...
1,4       1,4
1,46
1,6        1,6
1,66
1,8         1,8
...
3,0          3,0
3,06   
               3,2
3,26
               3,4
3,42     
               3,6
3,69
                4
                 4,2
4,21
                 4,4
4,47    
                 4,6
4,74
                  4,8
5,0              5,0
5,2              5,2
5,26
                   5,4
5,46
5,6              5,6
5,8              5,8
6,0              6,0
....

While the layer thickness for the structure extrusion is constantly the defined 0.2 mm, the thickness variation of the support extrusion is extremely variable. Especially for the very thin layers (0.06mm) the extrusion shift is often smaller than the resolution of the extruder, so it makes definitely no sense to run these layers. Furthermore it seems to me, that this makes an overall degradation of the 3D print. Is there any possibility to harmonize these layers with the main structure?

e-u-g, 2016-02-04:
and more...
it leads to underextrusion at start perimeters of main structure.
i guess, switching from microlayer support to a basic layer main structure(0.06 > 0.2) causes too sharp flow increase. it's big problem for any extruders.
this is not visible when printing from the innermost to the outermost perimeters.
but if option "external perimeters first" is enabled, then it have defect surface.
and this microlayering in support is random, no logic and dependences.
it is old problem in older versions also.
might somehow disable this microlayering?

lordofhyphens, 2016-03-15:
Hey, could you upload a copy of the config, a model that this issue appears on, and a screencap from Slic3r. The generated gcode might be helpful, but not as likely if you can reliably cause the issue from model+config. 

If you can't post the whole model, see if you can use the cut operator to just grab the part of the model that the support would generate for.