slic3r/Slic3r #3207: Post-processing script not working when output file isn't last argument, or parameters!

CleverCoder, 2016-01-16:
I'm trying to run a branch of GPX that allows outputting to serial port. The command I'm using looks like this:
C:\Data\Projects\GPX-markwal\master\gpx.exe -m fcp -u -v -s [input_filename_base].gcode COM3

This works fine from the command-line, but Slic3r seems to reject the command (probably beacuse it's sticking the output filename at the end), complaining it's not executable.

In fact, passing any arguments to gpx.exe seems to result in the same problem. I suspect that I need to wrap my command inside of a batch file, but that's kinda lame.

markwal, 2016-01-16:
It's been this way for a while, you really can't have the box have anything in it since slic3r automatically appends the input file name.

Here's what you do:  Put only C:\Data\Projects\GPX-markwal\master\gpx.exe in the box (and it has to be a full path including the extension even if on path, because slic3r wants to check that it exists and is executable).

Then put a gpx.ini in that same folder with gpx.exe with the settings you're looking for.

However, you won't be able to get serial output that way, you'll have to wrap it in a batch file or python script or something to pull that off.  However, I don't think you want to do that.  Even though my GPX does talk serial, it lacks all of the controls of a good host.  For that, you should use OctoPrint with the OctoPrint-GPX plugin.  Then you get the host and the gcode translation.

CleverCoder, 2016-01-16:
Mark,

Thank you for the tips! I found that calling from a batch file (With %1 for
the filename) does work pretty well. Thanks for the advice. I'll try the
Octoprint-based method tonight.

Interestingly, I think I was in the middle of drafting an email to you
around the Windows 10/COM port issues that seem to plague GPX.exe, and
other slicers. I was convinced I had a fix for it earlier today, but then
things seem to have went South at some point. (BTW, I'm a software engineer
by day - albeit a little rusty in C) I've been doing battle with USB based
printing since I got the printer. Long story short, I was hacking the GPX
source to tinker with every possible setting under the sun, including using
different API's (such as WriteFile), tweaking buffers, COM settings, you
name it! Looks like Simplify3D is also plagued with the same problem. RepG
seems to work fine, but I don't have the fan control I need. Have you seen
the issues with Windows 10 USB/COM stack by chance? I'm pretty determined
to figure out what's going on, since it seems like such a common issue. It
would be great to get it working. Also, the other slicers that work OK over
USB are MakerWare (however, newer versions don't support the second
extruder!), and Matter Control (I need to spend more time tinkering with
GCode to get my FFCP to work). MakerWare is closed source (AFAIK), and
MatterControl is based off .NET.

Anyway...again, I appreciate the reply, and the advice.

Cheers!
- Sean

markwal, 2016-01-17:
You might try picking up a raspberry pi and run OctoPrint on that. Then you upload gcode via the browser to the pi and then serial from the pi to the bot. Works nicely and slic3r even has a send to printer button that uploads the gcode directly.

People haven't much luck running OctoPrint-GPX on Windows.

CleverCoder, 2016-01-17:
Perfect timing! I've actually been experimenting with OctoPrint.... from
Windows. I have everything working, except I can't seem to get
OctoPrint-GPX to build, even under MINGW. It seems to continue to want to
use the MSVC compiler (specifically, the version 9 compiler I installed for
Python), rather than GCC. I have the GPX export running well, dumping to my
SD, with fcp machine profile. It works beautifully for that. I might come
back later (when I regain the patience) and take another stab at building
OctoPrint GPX under Windows. I'll probably go ahead and pickup a Pi device.
It seems like a good, proven solution.

Thanks again, and if you happen to hear of any breakthroughs with the USB
driver on Windows 10, I'd love to hear. Also, thanks again for the great
work.

- Sean

markwal, 2016-01-17:
I think there is cfg file for python's setuptools where you tell it you want to use gcc.

CleverCoder, 2016-01-17:
Defeated. After spending over 4 hours on this thing tonight, I just gave in
and ordered a Raspberry Pi 2. I tell ya.. What a painful adventure.
Installing MSYS2, pulling down all sorts of toolchains. I eventually got
the extension to compile with gcc 64-bit (under 'old' MinGW with 64-bit
tools in PATH), but then it barfed on the "Using GPX to connect" or
whatever. My main struggle was to get the 64-bit compiler to work, as it
seems the linker threw an exception when I tried building with 32-bit.
Perhaps it's because I was on 64-bit Windows. By the way, the tip about the
compiler for python to use was spot-on. I basically had to perform a
"python start.py build" under MingW, and then under the Windows "Virtual
Environment" run the 'install' portion. It registered.. it showed up. I was
excited. Maybe another day.

Thanks again for the feedback and pointers. One day, maybe we can get this
GPX business working over USB on Windows 10. (Of course, this is nothing to
do with Slic3r, but I'm always up for a good challenge, and this USB COM
stuff is my nemesis!!)

Cheers!
- Sean

alexrj, 2017-03-16:
No arguments are allowed for post-processing scripts for security (people exchange config.ini files easily and we don't want to supply anything to a shell). The recommended workaround is the one you found: write a .bat or .sh wrapper script.

closed 2017-03-16 by alexrj