slic3r/Slic3r #3210: Solid fill is not calculated correctly

mkubi, 2016-01-18:
[image]
[image]

Hello

seems like in some cases the solid fill is not calculated correctly (please check Slic3r_bug.png). The issue disappeared when I rotated model by 90 degree (Slic3r_correct.png). Please check attachement.

source file : http://www.thingiverse.com/thing:1102368 (Prusa_i3_RAMBo_V3_bottom_plate)

regards,
Marian

lordofhyphens, 2016-01-18:
Please read the guidelines, we need the configuration you used at least.

mkubi, 2016-01-18:
[config.zip](https://github.com/alexrj/Slic3r/files/94539/config.zip)

sorry my fauld. Configuration attached.

Vicious-one, 2016-01-22:
Same thing here. In fact, the part where infill seems less dense is heavily overextruded.

mkubi, 2016-01-22:
Seems like overextrusion is applied also to rotated object which I originally thought is good. So far I only noticed the issue just with this particular object.

mdelongpre, 2016-01-22:
Add me to the list, no matter what I do it seems to overextrude for me as well

lordofhyphens, 2016-07-12:
I've no idea why it is happening, but I found that if you set the perimeter width to 0.4 (from 0.45) the symptoms go away... but also when you have 1 0.9mm perimeter as well, so it's not the space available doing it apparently. 
[image]

It's ignoring the solid infill setting completely too -.-;

bubnikv, 2016-07-12:
This issue certainly has a high priority. I am on vacation this week,
having a phone only. I will look into this the next week. If you could only
assign it to me😉. There is a logic in the infill code to stretch the
extrusion width to fill a top/bottom surface with an integer number of
lines, but that code shall not have this effect.

<https://cloud.githubusercontent.com/assets/31754/16753030/d286027e-47ab-11e6-8fcb-4348942fd743.png>

<https://github.com/alexrj/Slic3r/issues/3210#issuecomment-231935127>, or mute
the thread

alexrj, 2017-03-05:
This is fixed as of current master:

[image]

closed 2017-03-05 by alexrj