slic3r/Slic3r #3212: Issue: Max extrusions speed has preference over retract speed

frank16, 2016-01-19:
This happens at least with repetier firmware in slice3r 1.2.9
In eeprom I can set a maximum extrusion speed (something like 2mm/s on 3mm abs on my K8200)
Retraction speed is set at 40mm/s, but it will retract at no greater speed then what I set for max extrusion, so very very slow.

Automatic extrusion in repetier firmware is set to off, because that has another bug, it will not retract at all, just push, but, at the proper speed :-(

Reasonably speaking this is a bug of slic3r.

xoan, 2016-01-19:
Can you paste some lines of a gcode file generated by Slic3r along with your config?

It seems that this is a firmware issue not related in any case with Slic3r.

frank16, 2016-01-22:
I'm trying hard to figure this out. Could you explain what is used by slic3r (I'm using repetier host and repetier firmware) for configuration?
Obviously the slic3r config files, the printer configuration is passed, but not the eeprom settings?
Is it possible that repetier host changes the generated G-code on the fly using eeprom settings? It probably does for speed and extrusion, which you can set manually on the fly, but for limit values set in eeprom?
It's confusing :-(

xoan, 2016-01-22:
Open Slic3r and "File → Export config..." (not bundle)

lordofhyphens, 2016-05-29:
Why are you expecting Slic3r to override your firmware? It's not supposed to do that. If you really wanted to do that, write a short postprocess script to set and reset the max E axis feedrate before and after each retraction. The retraction sections should be called out with verbose gcode.

Or use volumetric extrusion (I don't know how to turn it on in repetier fw) and set the volume limit separately from the axis limit.

Closing.

lordofhyphens, 2016-06-02:
I'll reopen this and add it to "PR or bust" if someone wants to supply a post-process script to do what @frank16 wants.