slic3r/Slic3r #3215: Preview shows infill overlapping perimeters on one layer - caused print fail?

mattster98, 2016-01-21:
Started happening with one particular model.  Printed it once just fine, even though I noticed the weirdness in the preview.  Next time I re-sliced it to try and make it go away, but it was still there, just different.  Caused the print to shift on one axis at the layer with the weirdness.  Confirmed the thickness of the pre-fail resulting print is 5.6mm - same as the layer where the preview shows strange infill.

I checked the STL in netfabb and it didn't find any problems.

Here's how it looks on the preview:
[image]

And the resulting prints (the one in white I noticed and killed, then pulled off most of the messed up part):
[image]
[image]

Settings file - refer to ones with "weird_infill" suffix...  I had been tweaking trying to make this go away.
[weird_infill_settings.ini.txt](https://github.com/alexrj/Slic3r/files/99159/weird_infill_settings.ini.txt)

STL File:  http://www.thingiverse.com/thing:872507

Resulting gcode: 
[Cutter_Tool_Holder-weird-infill.zip](https://github.com/alexrj/Slic3r/files/99162/Cutter_Tool_Holder-weird-infill.zip)

lordofhyphens, 2016-01-21:
Try setting Gap Fill Speed = 0 and check the output, or setting your infill width to some fixed value. 

I've been wondering where this could live myself.

lordofhyphens, 2016-01-21:
I wonder if this is related to #3172

mattster98, 2016-01-22:
Changing Gap Fill Speed to 0 didn't change anything (that I could tell from the preview).  Infill extrusion width was 0.45.  Tried changing that to 0 and I think that made it worse.

[image]

Tried changing the default extrusion width to 0 (along with infill width) and it seems to have helped.  Still some sort of visible artifact at 5.6mm height.

[image]

mattster98, 2016-03-25:
I may have run into another instance of this.  Doesn't have the same visual indication on the preview, but several attempts at printing this one model with changes to settings each time (and even changes to the model)  have yielded a failure similar to the above at a relatively similar Z height.  Makes the printer appear to have skipped steps, but other models print fine..

Is there a way to more easily hunt through the gcode to try and find the line(s) that may be the cause?  I'd love to be able to confirm if this is the explanation.

lordofhyphens, 2016-03-25:
Not really, finding overlaps like that is hard to do without being able to
visualize the 3d space.

mattster98, 2016-03-25:
Is there a visualizer that shows the gcode more sequentially rather than
based on Z-height?  gcode.ws, slic3r's built-in, and octoprint's built-in
are all I've used.

lordofhyphens, 2016-03-25:
Yagv perhaps.

alexrj, 2016-03-26:
This looks fixed as of current stable branch (1.2.10-dev):

[image]

I need to test more with 1.3.0-dev as it's either hanging or taking too much time. This is a very tricky model.

DrLex0, 2016-11-15:
I have another model that exhibits this problem very prominently. The infill breaks outside the perimeters and is excessively thick. I didn't even try to print this but I am sure it would have ended in disaster.
When this bug occurs, it is always when I either enforce a specific default extrusion width, or a specific perimeters width. If I leave both at 0 (auto), the problem goes away. Setting an external perimeter width does not seem to cause problems.

[image]
[image]
This looks like a delicious cake! What it should look like:
[image]

The model and settings:
[ModelAndConfig.zip](https://github.com/alexrj/Slic3r/files/592356/ModelAndConfig.zip)

lordofhyphens, 2016-11-15:
@DrLex0 have  you checked it with current stable branch?

DrLex0, 2016-11-15:
I have checked it with the latest Prusa3D build which does not have it (but as far as I understand from the release notes, they basically locked down some of the automatic infill parameters to circumvent this problem).
I haven't tried it with the official Slic3r branch because building it in OS X 10.11 was a royal pain the last time I tried it.

bubnikv, 2016-11-16:
> I have checked it with the latest Prusa3D build which does not have it (but as far as I understand from the release notes, they basically locked down some of the automatic infill parameters to circumvent this problem).

In the latest Prusa3D code, the automatic line width adjustment is enabled to some extent, limited to +20%.

Also a similar issue you experienced is caused by a bug in the "turn bottom surfaces to internal bridges" feature, which may in some cases produce extrusions outside the object.

alexrj, 2016-12-21:
Thank you @DrLex0, I fixed that.

@bubnikv, this was actually caused by `adjust_solid_spacing()` not being idempotent and increasing spacing by 20% each time it was called during a single `fill_surface()` run.

closed 2016-12-21 by alexrj