slic3r/Slic3r #3216: Bug: On some machines slic3r will only use one core, regardless of threads setting.

BrainSlugs83, 2016-01-24:
I've upped Slic3r to use 16 threads, and it still only uses 12-18% of my CPU (on a hexacore machine).

It seems that regardless of the number of threads that are set, I can never get it to use more than 16% of my CPU.  This makes slicing very slow for me even on models that aren't that complex.

(Total CPU usage is about 2% higher than whatever slic3r is taking, so nothing else is slowing me down.)

I've heard from other users (forums and digging on this issue) that they're able to see it take up 100% of their CPU, and from others that it can only run on one core. -- So, I'm thinking that this only happens on certain machines / environments?

Slic3r Version 1.2.9 (as packaged with Repetier-Host V1.6)
Windows 10 (x64), 16 GB RAM, SSD RAID Stripe, 3.2 GHz Hexacore (AMD Phenom II X6 1090T Black Edition), all drivers (chipset, etc.) installed and working.

If there's any additional details I can provide please, let me know.

lordofhyphens, 2016-01-24:
Copy of a model you see this effect and the slic3r configuration will help.

BrainSlugs83, 2016-01-24:
I don't think it's model specific, but I've uploaded two models here where the issue happens.  The fillenium_malcon one jumps up to 20 to 25% for less than second during slicing, but it slices pretty quickly (30 seconds ish?) so it's not a big deal.

The maelstrom one takes **minutes** to slice and is at a steady 18% the whole time.

[models & config.zip](https://github.com/alexrj/Slic3r/files/102172/models.config.zip)

alexrj, 2017-03-18:
The slicing process needs to be single-threaded in some parts, so that's what you might be seeing. Anyway, we've been moving parallelization to C++ threads and doing much optimization work on that so the situation is probably much better than before, and continues to improve.

closed 2017-03-18 by alexrj