slic3r/Slic3r #3217: [feature request] add option for a filename dialog when sending to Octoprint

Azlinon, 2016-01-24:
When I send gcode direct to Octoprint, I don't see a way to specify the filename to use.  Instead, one of the models I added to the plater serves as the filename.  This is a problem because sending a PLA model to a printer loaded with ABS is bad (stripping the filament as a best case?), and then I also don't know what options were used or what's actually in the gcode.  Popping up a dialog box and then using that name would be ideal, IMO.

lordofhyphens, 2016-01-24:
The filename is currently in the print options and uses variables.

pyrophreek, 2016-01-25:
Is the filament type a variable? haven't been able to get that to work.  Had to create a print profile for each type of filament to be able to add that manually.  

Have to say I think this is a good idea, and for those who maybe don't want it, add an option to the output section or global preferences to enable/disable.

lordofhyphens, 2016-01-25:
You can use the filament preset name as a variable.

bj0, 2016-07-15:
Something like this would be very useful because, as is, if I try a print and then change a setting (like creating multiple copies of an object) and resend, octoprint's old copy gets overwritten.

alexrj, 2016-12-17:
Done! Slic3r now prompts with the filename and asks whether to start printing.

closed 2016-12-17 by alexrj

Azlinon, 2016-12-17:
Awesome!  Thank you.