slic3r/Slic3r #3220: [1.3.0-dev] Nonzero Max Volumetric Speed causing negative feedrates even if no automatic speeds are used

lordofhyphens, 2016-01-26:
Found this one, similar manifestation as #3045, except set max volumetric and make sure all other speed settings are nonzero.

alexrj, 2016-03-18:
How to reproduce?

lordofhyphens, 2016-03-18:
I'm honestly not sure anymore; I'll see if I can get it to pop up under current -dev.

JBKingdon, 2016-03-24:
I think I hit this today with master of Mar 19th. g-code is making very slow "printing" moves with negative feedrate and negative extrusion:

G1 X66.344 Y111.665 F12000.000 ; move to first infill point
G1 F-582474
G1 X66.142 Y111.941 E20.38560 ; infill
G1 X65.857 Y112.405 E20.38255 ; infill
G1 X65.777 Y113.020 E20.37909 ; infill
G1 X65.751 Y113.631 E20.37567 ; infill
G1 X65.750 Y130.364 E20.28220 ; infill
G1 X65.765 Y130.825 E20.27963 ; infill
G1 X65.813 Y131.316 E20.27687 ; infill
G1 X65.967 Y132.141 E20.27218 ; infill
G1 X66.103 Y132.638 E20.26930 ; infill

[stl-gcode-config.zip](https://github.com/alexrj/Slic3r/files/188427/stl-gcode-config.zip)

JBKingdon, 2016-03-24:
Ah, it seems to be more complicated than I thought. Setting max volumetric to 0 got rid of the negative F values, but I'm still seeing bizarre negative extrusion infill moves

G1 X74.181 Y141.190 F12000.000 ; move inwards before travel
G1 X74.772 Y140.725 F12000.000 ; move to first infill point
G1 F1200
G1 X74.971 Y140.865 E17.58194 ; infill
G1 X75.452 Y141.150 E17.57882 ; infill
G1 X76.036 Y141.224 E17.57553 ; infill
G1 X76.629 Y141.249 E17.57221 ; infill
G1 X93.361 Y141.250 E17.47875 ; infill
G1 X94.165 Y141.205 E17.47425 ; infill
G1 X94.686 Y141.128 E17.47131 ; infill
G1 X95.050 Y140.890 E17.46888 ; infill
G1 X95.446 Y140.590 E17.46611 ; infill
G1 X97.104 Y140.033 F12000.000 ; move to first infill point

lordofhyphens, 2016-03-27:
@JBKingdon is gap fill on?

JBKingdon, 2016-03-27:
Since gapfill doesn't have an on/off setting in the UI I'm never 100% certain if it's active or not. The full config should be in the zip file.

lordofhyphens, 2016-03-27:
@JBKingdon  setting its speed to 0 turns it off. It's a bit weird and I think it could stand to have better UI.

Currently the workaround I have is a short script that changes all negative feedrates to some constant (small) value.

JBKingdon, 2016-03-27:
So, to rehash a question from a previous thread, what happens when you are running autospeed and all (non-auto) speeds have to be set to 0 - is gapfill on or off then?

The work-around sounds interesting, but it's not just negative feed rates - in my second example all feedrates are positive, but now we're getting negative extrusion.

lordofhyphens, 2016-03-28:
With volumetric extrusion on and gapfill on, I'm seeing the negative
extrusion rates. Disabling gapfill (setting its speed to 0) removes the
symptoms.

lordofhyphens, 2016-05-27:
May be related, but I added a call to CONFESS if the F feedrate (what is actually written) is <= 0. Here's the output. 

Error in function _extrude at src/libslic3r/GCode.cpp:569: Trying to write a nonpositive speed!
	 at /usr/share/perl5/Slic3r/Print/GCode.pm line 610 thread 7.
	Slic3r::Print::GCode::_extrude_infill(Slic3r::Print::GCode=HASH(0x7b04fd0), HASH(0x7f3c1d9e6f90)) called at /usr/share/perl5/Slic3r/Print/GCode.pm line 563 thread 7
	Slic3r::Print::GCode::process_layer(Slic3r::Print::GCode=HASH(0x7b04fd0), Slic3r::Layer::Ref=SCALAR(0x7f3c1d9e6b58), ARRAY(0x7f3c1d9e6558)) called at /usr/share/perl5/Slic3r/Print/GCode.pm line 293 thread 7
	Slic3r::Print::GCode::export(Slic3r::Print::GCode=HASH(0x7b04fd0)) called at /usr/share/perl5/Slic3r/Print.pm line 405 thread 7
	Slic3r::Print::write_gcode(Slic3r::Print=SCALAR(0x7aa0940), "/home/lenox/3d/e3d_cooler/Cooling_Fan_Duct_for_E3D_All_Metal_"...) called at /usr/share/perl5/Slic3r/Print.pm line 75 thread 7
	Slic3r::Print::export_gcode(Slic3r::Print=SCALAR(0x7aa0940), "output_file", "/home/lenox/3d/e3d_cooler/Cooling_Fan_Duct_for_E3D_All_Metal_"...) called at /usr/share/perl5/Slic3r/GUI/Plater.pm line 1211 thread 7
	eval {...} called at /usr/share/perl5/Slic3r/GUI/Plater.pm line 1210 thread 7
	Slic3r::GUI::Plater::__ANON__() called at /usr/share/perl5/Slic3r.pm line 108 thread 7
	Slic3r::__ANON__() called at /usr/share/perl5/Slic3r.pm line 109 thread 7
	eval {...} called at /usr/share/perl5/Slic3r.pm line 109 thread 7

STL/config.ini where it breaks :)
[failure_case.zip](https://github.com/alexrj/Slic3r/files/286980/failure_case.zip)

alexrj, 2016-07-10:
@lordofhyphens, I was able to reproduce this with your failure_case.zip - good

alexrj, 2016-07-10:
I fixed this in current master. Thank you.

closed 2016-07-10 by alexrj