slic3r/Slic3r #3221: Slic3r fills area that should be empty space

justbuchanan, 2016-01-26:
I'm not entirely sure what the issue is, but for a particular item I tried to slice and print, slic3r seems to have filled areas that should be empty, then left the solid parts unfilled.  Almost looks like it's filling the negative of what it's supposed to.  Here's a screenshot of the item in Repetier Host:
[image]

And a screenshot of the "rendered gcode" after running slic3r:
[image]

The `stl` part file can be found on thingiverse [here](http://www.thingiverse.com/download:1045227).

Please let me know if there is anything I can do to help debug this further.  Thanks for the excellent software!

lordofhyphens, 2016-01-26:
First note - Rephost has no clue what the width of the filament should be.
Any "screenshots" of the path should use slic3r's own preview (which does
know).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

justbuchanan, 2016-01-26:
Thanks for the quick reply!

Here's a preview of the part in slic3r:
[image]

And here's a screenshot of the layer's tab showing that slic3r is filling in the area that should be blank.
[image]

It's possible that this is a problem with the stl file itself.  I'm not sure what it would be though because it looks fine in the viewer.

justmyopinion, 2016-01-27:
Your model is is not MANIFOLD. 
SLIC3R is reporting Auto repair of 23671 errors!
You cannot expect to get deasant result out of that. Thingiverse is not a good place to find quality models.. Check and repair STL file with Netfabb before slicing

justbuchanan, 2016-01-28:
Thanks for the info!  I figured it was probably the model file since slic3r has done a great job with many other stl files I've tried.  I didnt' notice the manifold errors before, but that explains a lot.

closed 2016-01-28 by justbuchanan