slic3r/Slic3r #3224: "3D Tab" Works well, "Preview Tab" and "Layers Tab" doesn't works well

mpp2user, 2016-01-28:
Hi,
I try to expose my problem:
1 - to load .stl file (see attached)
2 - select "3D Tab" :  works perfectly (see attached)
3 - select "2D Tab" :  works perfectly (see attached)
4 - select "Preview Tab" : the top part is not displayed (see attached)
5 - select "Layers Tab" and slide it : the top part is not displayed

the stl file should be ok ("Cura" works perfectly)

thanks in advance for your response
[Problem.zip](https://github.com/alexrj/Slic3r/files/108504/Problem.zip)

SFS87, 2016-06-10:
I'm having the same exact problem. I'm running Windows 10 with the latest version 1.2.9 and cannot see the slicing paths.

bubnikv, 2016-06-24:
Yeah, my Slic3r 1.3.0-dev hangs on the 3D preview.

lordofhyphens, 2017-03-28:
The provided STL has multiple parts in it. 

The model on the 3d preview does not look right to me: 
[image]

Here's the model after a split operation:
[image]

lordofhyphens, 2017-03-28:
[image]

This is the 3d model after it got passed through service.netfabb.com

And here's the 3d preview.
[image]

It looked like there were multiple intersecting shells. So each piece was manifold.

lordofhyphens, 2017-03-28:
1.3.0-dev doesn't hang anymore though.

xoan, 2017-03-28:
Even model is manifold, it has normal orientation issues.

[image]

All darkgray faces are inverted.

lordofhyphens, 2017-03-28:
Fixed model: https://service.netfabb.com/upload/download.php/Object_fixed.stl?job=3bfdab7213d411e7b4115254a20161ba

closed 2017-03-28 by lordofhyphens