slic3r/Slic3r #3225: Feature Request - Disable Cross-Hatching Infill, and new Parallel Infill Pattern

tiagojsalves, 2016-01-28:
Hello all. My name is Tiago, and I'm currently a Master student trying to optimize printing parameters to several filament materials, and I'm using Slic3r for the G-code generation.
As I am trying to create standard dog-bone test samples to test the material's mechanical properties, I came across a bit of a problem - I can't have cross-hatching infill, because to test the material to tensile stress, I need to have all infill aligned by the direction of tensile stress pull. By this, I mean that all lines of the infill (except near the perimeters) need to be aligned by the X axis, for example. 
You could even create a new infill pattern where the lines of the infill do not cross between layers, but instead, they could fill the gaps of infill lines beneath the active layer. This is a new idea of mine, a kind of Parallel Infill where all lines of the infill are parallel, at a given fill angle. This way you could get high density, with aligned inner structure.

For this same study, at the same time, I'll also need cross-hatched infill, rectilinear and others, but I also need several types of infill without being cross-hatched.
Besides trying to request a new feature to the program, I would like to ask you to indicate me a fast way to do this, by tweaking something, or by inputting some code in the command line, or by some sort of script of any kind.

Please, be kind to treat this issue as **urgent**, as for me and my colleagues, this is an real life urgent matter.

Best Regards,
Tiago Alves, from Portugal

VanessaE, 2016-01-28:
Set your infill pattern to "concentric" and you should get parallel lines along the majority of the object, assuming your dog bone is the "typical" shape.

lordofhyphens, 2016-01-28:
Seeing as the author basically works on this in his spare time, "urgent" means nothing. You will probably have to modify code yourself for the infill to get the extra things you want. Welcome to being a grad student.

lordofhyphens, 2016-01-29:
The infill is still done in Perl, so you can probably modify it to suit your needs.
https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Fill.pm

a4jp-com, 2016-01-29:
Nothing has happened with this software since December 26th and only Alex can update the files that get downloaded here so I don't think there will be a fix for you soon but you can download the files and muck around with the code if you like. 

When the thin wall code is turned on you will also get an overfill problem so make sure that is turned off until it can be fixed.

Can you post a picture of the infill you want? Please draw it and then upload the picture here.

Are you able to compile the code here by yourself you do you need help with that?

What type of system are you running?

tiagojsalves, 2016-02-01:
Hello all again.
First of all, I would like to thank you for posting your ideas.
Actually I made two separate requests:

1. The ability to disable the cross-hatching of any pattern (easier said than done, right?)
        -If you have any idea of how to disable this, please let me know.

2. New infill type (see comment below)

@VanessaE , for the case of the typical dog bone shape for tensile stress testing, the "concentric" infill should do the trick, specifically because on the test area (the narrower area of the shape), all lines are parallel. I haven't thought of it like that. Thanks a lot for that idea. For this specific test, that will workout just fine. Unfortunately, for square shapes (ex.: drop impact test), that won't work.
@lordofhyphens , in fact I could modify it, if I knew how. I'm actually a Product Design Engineer. I know a lot of things, but I don't know how to code Perl.
@a4jp-com , I actually designed the infill in 3D to show how it would turn out. In the image below, the layer height difference can be expressed as a function of the nozzle diameter:  [(√3)/2]*∅. The horizontal difference between layers would be half the diameter (∅/2). The great benefit is that, theoretically I would get 90% of material density, without accounting for the buckling of the filament, which would certainly happen during extrusion, achieving densities greater than this value. You can compare this density value with the Rectilinear, which would reach about 78,5%, theoretically speaking. Also, all filaments would be aligned perfectly to the tensile load, improving mechanical resistance, as I previously mentioned. 

[image]

Would you think this can be done? Is it feasible?

lordofhyphens, 2016-02-01:
You're a grad student, it's your job to work out how to do whatever you
need to do to get the research done :). I know because I've been a grad
student.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

a4jp-com, 2016-02-01:
You can rotate the infill to 90 degrees instead of 45 and then if the plastic is hot enough it will fill in the gaps so you don't actually have to have a pattern like that. If the pieces of plastic were just sitting on top of each other that pattern would be useful but as the plastic bonds to itself I can't see why you would offset every second layer to half the width of one printed line. 

How about actually drawing all the internal bits (structure) in 3D and then slicing it? That way you can perfectly control everything.

justmyopinion, 2016-02-02:
Maybe take a look at this and understand how printer is making a solid filling:
[http://manual.slic3r.org/advanced/flow-math](url)

tiagojsalves, 2016-02-02:
Any ideas on how to disable the infill cross-hatching?

a4jp-com, 2016-02-02:
Easy.

Print Settings > Layers and perimeters > Horizontal shells: 0 0
Print Settings > Infill > Advanced > Fill angle: 0

Fill pattern: Rectilinear 
100%

tiagojsalves, 2016-02-03:
Not that easy. 
Still crosses one layer after the other. The filaments are not parallel throughout the entire thickness, as in the image I shown previously. Maybe there are some other options you have selected that I didn't.

xoan, 2016-02-03:
I haven't any Perl knowledge, but I think you can take a look into Slic3r::Fill::Base code at https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Fill/Base.pm (I guess the most easy hack could be changing "PI/2" by "0" in both angles() and $rotate[0])

tiagojsalves, 2016-02-03:
Moitas grazas Xoan.
I'll try to do that, and I'll check if it solves the issue of cross-hatching. 
I'll try to learn how to change it, and recompiling the source code myself.

I'll let you know if it solved the issue.

Again, thank you very much.

antigpas, 2016-03-15:
Hey there,nice topic and actually I know a lot of people who want this feature disabled..Anyway, I tried to change these values that xoan suggested in the Slic3r::Fill::Base code, and loaded the script as a post-processing script in Slic3r, but in the Plater settings-preview mode there's still cross-hatching. Is this the right way to do it? I mean should it be loaded as a post-processing script in Slic3r? And also, when you load such a script in Slic3r and you export the g-code, should it be visible in the preview mode or only in the printer control software (i.e. pronterface etc...)? Thanks!

lordofhyphens, 2016-03-15:
I don't think that the fill pattern can be changed as a post-processing
script (because post-processing scripts work on the output from the slicer.
You'd need to add a new option so that it was generated.

Except for trivial cases, I don't know how you could undo the fill
crosshatch in a post-processing script.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

antigpas, 2016-03-15:
Thank you! Ok I see what you mean. So I need to change something in the software itself, right? If yes, could you recommend any tutorial/guide online  (not on Slicer in particular) on how to do so? (PS be kind, I'm amateur on programming or so), thanks

lordofhyphens, 2016-03-17:
Since fill is done in Perl, you should probably develop the ability to hack
around in Perl and read code. Honesty, the best way is, IMHO, to look for
something that looks like it governs what you want and start changing
things until you can work out what goes where.

It won't give you code that you could submit to slic3r as a pull request,
but it probably would let you get by as a grad student trying to get their
thesis done

I say this as a grad student.

bubnikv, 2016-09-23:
I wonder what is the purpose of having all infill lines oriented the same way? Maybe optical purposes? Other than that, I cannot imagine this will make the part stronger.

lordofhyphens, 2016-09-23:
@bubnikv He is trying to do something for his research. I've been a grad
student; I am fairly familiar with the routine. I do not think this is
something that should be worked on by us to support someone's grad
research, especially when we can't see a benefit for doing infill weirdly
like this.

bubnikv, 2016-09-23:
I know Simplify3D supports something like this. But simply having the option does not mean one shall use it.

In Simplify3D, the various infills are configured through a dialog by repeating a pattern multiple times per layer with configurable rotation angles.

forrestp, 2016-09-23:
The main reason for using this infill pattern is for testing new materials - stuff with strands of fibers or optics, etc. Also, it can be useful for characterizing adhesion within a single layer. 

Overall, this probably isn't a very useful feature for most people. However, I needed it and decided to implement it. After seeing this thread and a couple other people looking for this (@tiagojsalves, @antigpas) I figured I'd make a pull request to open up a discussion about adding this as an infill option.

Even if we decide not to merge this,  I wanted to make it easy for people looking for this feature in the future to find my fork.

lordofhyphens, 2016-09-23:
No problem, thanks. The code itself looks good.

From a UI perspective I am wondering if it is better to instead have
control over the angles used. As @bubnikv mentioned, that was the approach
that S3D used. For the immediate purpose both approaches are valid.

bubnikv, 2016-09-24:
The problem with the way Simplify3D and Slic3r as well combine the two line
fills to do the rectilinear fill is that they possibly place two connecting
lines along the perimeter one over another, leading to an overfill. So a
smarter algorithm is actually desirable to avoid these overfills.

On another note, Slic3r connects the infill lines by straight segments,
while Simplify3D follows the perimeter.

PR #3518 (2016-11-22): Aligned rectilinear infill
Fixes #3225 with a new fill option called 'Aligned Rectilinear'
commit a3ddfde5 (2016-11-22): Add fill pattern for aligned rectilinear fixes #3225
closed 2016-11-22 by alexrj