slic3r/Slic3r #3227: git HEAD compile failure, "SESSION_MANAGER environment variable not defined"

VanessaE, 2016-01-29:
After multiple tries to get this thing to compile, I got all of the dependencies installed, or so I assume...  Building the command-line part using `perl Build.PL` seems to have worked, however, the compile of the gui part fails, apparently because of the aforementioned "SESSION_MANAGER" error.  I run Fluxbox as my window manager, and as far as I am aware, no session manager is actually running (nor have I been able to find out how to start one), but... well... why should I need that to compile a program? :-)

Relevant build log:  http://pastebin.ubuntu.com/14693084/

(Incidentally, it's unclear where "--force" should actually be added.  `perl Build.PL --gui --force` does not help.)

lordofhyphens, 2016-01-29:
For ubuntu I recommend grabbing the source ubuntu package from your distro, unpacking it and then overwriting the source with the HEAD. This lets you build a package which may/should sort out what's going on. Compiling by hand from the upstream stuff invokes CPAN, which tends to do weird things. 

I have a Debian branch already ready to go if you want to use that: https://github.com/lordofhyphens/Slic3r

VanessaE, 2016-01-29:
As a matter of fact, I am using Debian :-)

That said, your fork doesn't do so well.  After deleting ~/perl5 (I don't use --sudo), and starting from the beginning with the non-gui `perl Build.PL` step, the compile succeeds, but all tests fail:

http://pastebin.ubuntu.com/14693786/

(fwiw, I use the Ubuntu pastebin because it handles long lines well...ish).

lordofhyphens, 2016-01-29:
My fork is basically HEAD with the packaging stuff (originally from the
debian maintainer) added. From the checkout folder make sure you are set up
to build packages in debian (fakeroot, etc) and run "fakeroot debian/rules
binary". It worked on my Jessie system and it worked when I ran Wheezy.

lordofhyphens, 2016-01-29:
I will make a test build on one of my systems in the morning to check as
well.
On Jan 29, 2016 12:29 AM, "Joe Lenox" <lordofhyphens@gmail.com> wrote:

> My fork is basically HEAD with the packaging stuff (originally from the
> debian maintainer) added. From the checkout folder make sure you are set up
> to build packages in debian (fakeroot, etc) and run "fakeroot debian/rules
> binary". It worked on my Jessie system and it worked when I ran Wheezy.
>

VanessaE, 2016-01-29:
Ah HAH!  THIS worked (using your fork):
```bash
sudo apt-get build-dep slic3r
sudo apt-get -y install libdevel-checklib-perl libwx-glcanvas-perl
cd /home/vanessa/RepRap/Slic3r-git-lordofhyphens/ 
fakeroot debian/rules binary
dpkg -i ../slic3r_1.3.0-git+dfsg-1_amd64.deb
```
That's enough to get "1.3.0-dev" to run and display its gui.  Whether it behaves as designed is to be determined, but at least it finally built. :-)

lordofhyphens, 2016-01-30:
Glad I could help; could you close this then?

VanessaE, 2016-01-30:
I guess I should, yes. :-)

closed 2016-01-30 by VanessaE

lordofhyphens, 2016-01-30:
I try to keep the base fork updated to git HEAD on upstream, so you can continue to use it.

VanessaE, 2016-01-30:
Thanks, I appreciate it.