slic3r/Slic3r #3228: Slicing never ends.

skish, 2016-01-30:
Hello! Slicing with attached&zipped config and stl never ends. Mainfold: Yes.
1.2.9 both linux and windows. "Processing triangulate mesh..."
Thanks.

[src.zip](https://github.com/alexrj/Slic3r/files/110938/src.zip)

c-born, 2016-03-01:
I can confirm this, or a similar problem, having just come across it when trying to slice a very simple pot-plant support for my wife.

When I Add the object to Slic3r, it just sits and spins, until i have to kill the process. This happened with the older (1.2.9?) version I was running, and after I upgraded to the latest head the behaviour was the same.

Using my old favourite version 0.9.10b it slices quickly and with no problems.
(The only change required to use 0.9.10b is to change the fill_density from 40% to 0.4, as the older version chokes on the % sign) 

I'm using this on Windows 7 64

[PotPlantSupport.zip](https://github.com/alexrj/Slic3r/files/152094/PotPlantSupport.zip)

lordofhyphens, 2016-09-28:
[image]

I can't reproduce on 1.3.0-dev (3700950474583e6c64163fab848a831cdcfbf44e)

Give it a try - https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/view#

Going to close this, please post if you are still seeing problems.

closed 2016-09-28 by lordofhyphens