slic3r/Slic3r #3230: Infill every x layers limited output to 2

ausey00, 2016-02-04:
Using 1.2.9, setting 'infill every x layers' to a value of 2 or more, has no effect on the gcode produced. The infill is always produced as if this setting were set to 2.

If set to 1, the gcode output correctly reflects this setting and infills every layer.

xoan, 2016-02-04:
Please.., post your config and the STL file you are slicing, and also information about your system.

FYI, this setting is working for me in GNU/Linux and Slic3r 1.2.9

ausey00, 2016-02-04:
Apologies...
Here's the config i'm using. [Link to config.](https://www.dropbox.com/s/iz9gctkabbttmfj/Slic3r_config_bundle.ini?dl=0) The problem is still present if I disable the second extruder.

I'm using Win7 SP1 and Slic3r 1.2.9.

STL is a basic 20mm cube found here... [Link to STL](http://www.thingiverse.com/download:108498)

See screenshot, Where setting is set to infill every 5 layers, but sliced as if set to 2.

[image]

xoan, 2016-02-04:
The problem is that combine infill also takes into account the solid infill, and the max layer heigth is limited by the infill extruder and also the solid infil extruder (yo can see how it works if you choose Extruder 1 for both infill and solid infill). But now the problem is the top/bottom solid infill...

ausey00, 2016-02-04:
Ah, you're right. Is this expected behaviour then? 

In this case, as you can probably see from the config, I have a 1.2 nozzle; that intentionally has a high volumetric extrusion limit, and a 0.4 nozzle for detailed perimeters and top/bottom layers.

The intention was to do only internal infill at a high layer thickness, but this is limiting the ability to configure Slic3r to do so.

xoan, 2016-02-04:
I guess it is not the expected behaviour, but actualliy is the actual behaviour.

Solution could be adding a new parameter in multiple extrusion configuration to allow different extruder for solid infill and top/bottom solid infill; or change the behaviour of the actual perimeter extruder to something like "shells (both horizontal and vertical).

ausey00, 2016-02-08:
Agreed. Although as a short-term fix,  I'll probably just remove the number of layers combined safeguard in my own branch.

Thanks for your help xoan!

closed 2016-02-08 by ausey00