slic3r/Slic3r #3231: Where's Alessandro?

thiagopeixoto16, 2016-02-04:
No updates since last year, I'm afraid about it turned an abandoned project...
I really like Slic3r but there are still a lot of work to do to make it perfect.
I also know that is not easy to @alexrj manage it all but it will be good to see some activity here.
Since I am newbie on codding I can only wish better days :-)

xoan, 2016-02-04:
Last commit: 83c91a3538661af2447eb0e90489ec55c4a4d52e (26 Dec 2015)

lordofhyphens, 2016-02-04:
Development work happens in fits and spurts. so relax :)

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

a4jp-com, 2016-02-05:
He is working on something else at the moment but if we are lucky it might be part of Slic3r.

https://github.com/alexrj/DBIx-Lite

a4jp-com, 2016-02-15:
I hope he comes back soon though. Nothing has happened here for 2 months now.

lordofhyphens, 2016-03-13:
@alexrj is alive and looking through stuff. Closing!

closed 2016-03-13 by lordofhyphens

alexrj, 2016-03-13:
Yes, sorry everybody. I'm unhappy about the little time I had for Slic3r. I really wish someone could start helping me on coding/porting...

a4jp-com, 2016-03-14:
Is there anything I can help with on the testing side?

Lenbok, 2016-03-17:
@alexrj Even if it's not directly helping with coding, it is really great to see you delegating issue triage to others like @lordofhyphens. Maybe it is also worth creating a gitter chat room for Slic3r so that people can coordinate?

lordofhyphens, 2016-03-17:
I am usually lurking in #slic3r in freenode as well as #reprap as LoH

markwal, 2016-03-21:
@alexrj I'd like to lend a hand.  I was dorking around with the thin walls problem after I saw your message here.  I had gone down the route of attempting to use the distance from the retained half-edge to the line segment that caused its voronoi cell and then doubling it for the width, but you beat me to a solution (three solutions, all better than where I was :-)

So, if you want, perhaps I should do a PR on some porting.  Maybe I should start small so you can give feedback.  Where would you like me to start?