slic3r/Slic3r #3233: Zooming without mousewheel

szalat, 2016-02-05:
I have not found the possibility of zoom without using the mouse wheel in Slic3r interface. I think it would be useful for people who use a laptop with a trackpoint (eg. Lenovo) or without traditional mouse.

It would be useful to add the possibility of zoom via keyboard (eg. Ctrl - + / ctrl--) or with mouse and a key mod (eg. Ctrl, Shift).

Im using:
Linux 4.3.0-1-amd64 #1 SMP Debian 4.3.3-7 (2016-01-19) x86_64 GNU/Linux
and Slic3r 1.2.9

a4jp-com, 2016-02-05:
Nice point.

+1

If the plus and minus keys aren't being used for anything then just using them alone would be even better. Most keyboards have a number pad as well.

VanessaE, 2016-06-01:
As long as we're on the subject of keyboard control, Blender lets you press keypad 1, 3, or 7 (perhaps with Ctrl) to align the virtual camera to a given axis.  

Having that same feature in Slic3r's 3D and Preview panes would be very useful.

PR #3864 (2017-04-29): Adding Zoom
Fixes #3233
closed 2017-04-29 by alexrj