slic3r/Slic3r #3234: Slic3r 1.2.9 crashing in OSX 10.11.3

eagleapex, 2016-02-08:
Slic3r is crashing during slicing. This is during "background processing" or normal slicing. Manifold STls used. I can't think of anything to troubleshoot.

xoan, 2016-02-08:
I think you need to read this: https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

eagleapex, 2016-02-08:
What is there that I should read? Sorry. I'm on a train.

lordofhyphens, 2016-02-08:
It is the list of things to have to include in your issue to have it not be
ignored.

eagleapex, 2016-02-09:
Thanks lordoghyphens
Slic3r version 1.2.9
Mac OSX version 10.11.3
Config file and problem STL zipped and attached.
[config&stl.zip](https://github.com/alexrj/Slic3r/files/122530/config.stl.zip)

Expected result: Click  Export G-code, Slic3r slices and saves Gcode.
Actual result: Click Export G-code, slicing progresses to unknown point, slic3r window closes without error.

alexrj, 2016-03-18:
Hm, I can't reproduce this on OS X 10.8.5.

DrLex0, 2016-07-06:
I can confirm that slic3r crashes at seemingly random moments in 10.11.5. It is still usable, but pretty annoying of course. Generally, restarting it and trying the same thing again, will work, there is no specific recipe to reproduce this. The only steps to reproduce I can give, are: “load any STL file and fiddle with the settings to trigger background re-processing”. Sometimes it goes kablooie immediately after loading the first STL, sometimes I can work with it for 10 minutes.

As far as I can see, all the crash reports are similar and show a crash at “Perl_pp_entersub”.
Attached is one of the crash reports from ~/Library/Logs/DiagnosticReports/.

[slic3r_2016-07-07-005634_LexMBP17.txt](https://github.com/alexrj/Slic3r/files/351056/slic3r_2016-07-07-005634_LexMBP17.txt)

bubnikv, 2016-07-07:
@DrLex0 
Could you please try the Prusa3D MacOS build?
https://github.com/prusa3d/Slic3r/releases
Vojtech

lordofhyphens, 2016-07-07:
@DrLex0 yeah it'd be helpful to know if any of the series of modifications
@bubnikv has done has any effect :)

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

DrLex0, 2016-07-08:
I have tried the Prusa3D build, it looked promising while I tortured it with a pretty complex model, but after a while it also crashed. The crash logs seem somewhat different from the old ones, so maybe _something_ has changed :)
I also have the impression that it takes longer for it to crash, so my guess is that there were multiple causes for these crashes and some have been fixed.
[perl5.22.0_2016-07-08-012944_LexMBP17.txt](https://github.com/alexrj/Slic3r/files/353250/perl5.22.0_2016-07-08-012944_LexMBP17.txt)
[perl5.22.0_2016-07-08-021129_LexMBP17.txt](https://github.com/alexrj/Slic3r/files/353283/perl5.22.0_2016-07-08-021129_LexMBP17.txt)

lordofhyphens, 2016-07-08:
Is the model manifold?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

bubnikv, 2016-07-08:
Do you have the background processing on?

The way the background processing is implemented in Slic3r is suspicious at least. Often when a parameter is changed in the UI, the Perl background thread is being killed. This is never a good idea and I am not sure how Perl interpreter handles this. Then there are the C++ object instances, some owned by the UI thread, some of them shared between the UI and the background Perl threads. I will look into this issue one day.

Vojtech

bubnikv, 2016-07-08:
and by the way thanks for the crash logs. I believe they confirm my suspect of killing the background Perl threads. Both crashes look the same and it seems they happen at the end of a Perl background thread.

By the way, the Perl thread support is immature (it is year 2016) and due to the future of Perl the threads will never be modernized. To name the worst quirks: Starting a worker thread copies all the variables of the UI thread to the worker thread automatically, including all the UI. The wxPerl UI objects will have their internal pointers to the native objects zeroed when cloned to a background thread, but not the native Slic3r objects. I can go further, but believe me it is bad.

Thread 6 Crashed:
0   perl5.22.0                    	0x000000010823eef0 S_find_and_forget_pmops + 48
1   perl5.22.0                    	0x000000010823ef23 S_find_and_forget_pmops + 99
2   perl5.22.0                    	0x000000010823ef23 S_find_and_forget_pmops + 99
3   perl5.22.0                    	0x000000010823ef23 S_find_and_forget_pmops + 99
4   perl5.22.0                    	0x000000010823ef23 S_find_and_forget_pmops + 99
5   perl5.22.0                    	0x000000010823ef23 S_find_and_forget_pmops + 99
6   perl5.22.0                    	0x000000010823e7ba Perl_op_free + 234
7   perl5.22.0                    	0x000000010829d78b Perl_cv_undef_flags + 315
8   perl5.22.0                    	0x0000000108302424 Perl_sv_clear + 868
9   perl5.22.0                    	0x0000000108302f8f Perl_sv_free2 + 95
10  perl5.22.0                    	0x000000010826c167 Perl_gp_free + 183
11  perl5.22.0                    	0x000000010830275e Perl_sv_clear + 1694
12  perl5.22.0                    	0x0000000108302f8f Perl_sv_free2 + 95
13  perl5.22.0                    	0x000000010826c197 Perl_gp_free + 231
14  perl5.22.0                    	0x000000010830275e Perl_sv_clear + 1694
15  perl5.22.0                    	0x0000000108302f8f Perl_sv_free2 + 95
16  perl5.22.0                    	0x00000001082f740c Perl_sv_clean_all + 92
17  perl5.22.0                    	0x000000010825fe18 perl_destruct + 3912
18  threads.bundle                	0x0000000108668201 S_ithread_clear + 177
19  threads.bundle                	0x0000000108668029 S_ithread_free + 297
20  threads.bundle                	0x0000000108668a15 S_ithread_run + 1909
21  libsystem_pthread.dylib       	0x00007fff8aced99d _pthread_body + 131
22  libsystem_pthread.dylib       	0x00007fff8aced91a _pthread_start + 168
23  libsystem_pthread.dylib       	0x00007fff8aceb351 thread_start + 13

DrLex0, 2016-07-08:
The model I had the latest crashes with, is a quite detailed Skyrim helmet (4.3MB STL file), but I've also had crashes with a simple cube or sphere.
Background processing is enabled indeed. I tried disabling it, but then I don't get any previews unless I export the G-code, which is a bit cumbersome. I therefore usually leave it on when experimenting with settings (which I tend to do a lot because I'm still a 3D printing n00b). It would be nice to have a way to manually trigger a preview without having to pick an output file, like a button or menu shortcut…

lordofhyphens, 2016-07-08:
@DrLex0 Yeah it's the threading + background processing (which is probably one of the reasons that @alexrj started working on a C++ UI). I've been looking at a straightforward way to implement a preview button.

bubnikv, 2016-07-08:
I don't think a rewrite of the UI to C++ is necessary. Quite sufficient would be to re-write all the background slicing code to c++ threads and to have the Perl interpreter run in a single thread only. Simply said, ditch the Perl threads. And yeah, you can even debug a single threaded Perl application! One day we may ditch the Perl completely, but it seems quite a lot of work right now.

Another issue with the Slic3r background processing is the time it takes before a background thread is killed when processing large models. The Perl interpreter thread cannot be killed during the execution of an XS native call. The correct solution is to introduce a "cancel" flag in both the Perl worker thread code and the native code, check it periodically and exit the computation if the flag was set. This is the only clean solution how to get rid of the crashes. The background thread shall never be just killed, it shall finish itself and clean after itself. I highly suspect this is not happening right now. Frankly, I just don't quite understand what happens with all the native objects linked to Perl, when you kill the Perl thread, but it will likely not be what you want.

lordofhyphens, 2016-07-08:
@bubnikv I'm mostly guessing w/r/t any architecture changes @alexrj does :)
Agreed that the interaction could be done better.

DrLex0, 2016-07-31:
After using the 1.30.0 Prusa build for a while and trying 1.2.9 again, I can confirm that the former is _much_ more stable. 1.2.9 crashes on average once every 5 times background processing kicks in, 1.30.0 lasts so much longer I cannot even give a guesstimate.

alexrj, 2017-03-16:
I'm closing this one. We now have automated builds for MacOS X, and Slic3r is much more stable. Please reopen if you find more issues.

closed 2017-03-16 by alexrj

lordofhyphens, 2017-03-16:
@DrLex0 We have more stable builds now for OSX (automatic, see Readme.MD for a link). Please use those.