slic3r/Slic3r #3235: Slic3r messing up on extruder order

RepRapMe, 2016-02-09:
When slicing files consisting of more than 10 extruders (Repetier-Firmware supports up to 16 virtual extruders for color mixing) via Repetier-Host then Slic3r sorts the toolpath as follows: 0, 1, 10, 12, 13, 14, 15, 2, 3, 4, 5, 6, 7, 8, 9
The order should of course be 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15

This was not an issue in Slic3r 1.1.7

Virtox, 2016-02-20:
Will try to test it later, but looks like it's using the default lexical sort.
If you run the perl version, change line 542 of lib/Slic3r/Print/Gcode.pm
`my @extruders = sort keys %by_extruder;`
to
`my @extruders = sort {$a <=> $b} keys %by_extruder;`

And to skip the toolchange shortcut , comment out lines 543 - 551:
`       # if (@extruders > 1) {`
`        #     my $last_extruder_id = $self->_gcodegen->writer->extruder->id;`
`        #     if (exists $by_extruder{$last_extruder_id}) {`
`        #         @extruders = (`
`        #             $last_extruder_id,`
`        #             grep $_ != $last_extruder_id, @extruders,`
`        #         );`
`        #     }`
`        # }`

lordofhyphens, 2016-03-13:
@RepRapMe Could you provide a config file and an STL/AMF so we can easily test this?

alexrj, 2017-03-14:
@RepRapMe, thank you, I fixed this. The shortcut is actually there by design and it's a feature: it will save a toolchange. Can you share your use case and why the shortcut is unwanted?