slic3r/Slic3r #3236: Weird moves during retraction

RepRapMe, 2016-02-10:
This is probably the "Avoid crossing perimeters" acting up however there are no perimeters at this section.

Would have included both the source amf and the complete gcode but they're 50 and 14 Mb respectively...

[image]
[image]

lordofhyphens, 2016-03-15:
Still need a copy of the exported config. To get a smaller test case, try cutting the models 1mm or so above the point where you see the problem and export the plate as STL (use Slic3r). Hopefully that'll still trigger the problem and be small enough to zip and upload.

Which version of Slic3r is this?

SparkyCola, 2017-01-02:
I have the same issue in version 1.29. It seems to occur only with avoid crossing perimeters, but not without it and between two objects (haven't checked if all avoid crossing perimeter attempts make sense though, but there are some issues related to that).
Here are the configurations (in the G-CODE, hope that's enough), plus my STL-File, I'd also like to help with further tests (if required).
[ShoppingCartCoinChip.zip](https://github.com/alexrj/Slic3r/files/681403/ShoppingCartCoinChip.zip)

lordofhyphens, 2017-01-02:
@SparkyCola Please try with 1.3.0-dev.

alexxy, 2017-01-12:
I also see this. Some strange moves while retraction