slic3r/Slic3r #3237: Improve auto-arrange algorithm with SVGNest

triffid, 2016-02-11:
https://github.com/Jack000/SVGnest looks perfect for improving slic3r auto-arrange!

alexrj, 2016-03-13:
Definitely! It would be nice to have a C or C++ port... I can't merge JavaScript :-(
https://github.com/Jack000/SVGnest/issues/6

lordofhyphens, 2017-09-07:
Some considerations:

SVGnest doesn't have a termination clause; it keeps going until the user stops it. For our needs we would (probably) want to run some N iterations of the algorithm with each "Auto Arrange" click, removing the existing state if objects are added/removed/scaled/rotated by the user.

We'd also need to calculate the bounding boxes with holes to allow it to properly nest objects inside other objects.

triffid, 2017-09-07:
It's open source, I'm sure upstream would accept a patch to limit runs if
it doesn't already export a single iteration method

lordofhyphens, 2017-09-08:
@triffid we have to port/reimplement it anyway for Slic3r (as @alexrj noted, we're not merging javascript into Slic3r), so my comment was more of an note when implementing, not as an objection.

lordofhyphens, 2017-09-11:
One proposal I'd have is to set the minimum distance between parts to be 5mm more than the distance threshold to retract before move; this would be to avoid more artifacts on external walls. The extra 5mm is to ensure that even you set it to 0 prints wouldn't be right on top of each other.

lordofhyphens, 2017-09-11:
We'll also need to resolve the bounding box to closer in the shape if we want to be able to properly nest things in other things. Initially though I think it would be fine to leave them as opaque. 

Other notes: 
* Extra UI would be likely desired to lock rotation on individual items on the bed in Z (that is, the profile only changes on the XY plane). I don't think we should rotate the piece along the X or Y axes at all; most parts have the layer alignment as a strength requirement. 
* We'd need to be able to decouple copies from the main (individual rotate); I think that copies currently share rotates.

kallaballa, 2018-01-18:
I'm working on porting svgnest to C++: https://github.com/kallaballa/nestcpp/blob/master/src/util/geometry_util.cpp

At the moment i'm translating one function after the other. Anyone wants to join the effort?

FatemaKhalid, 2018-03-21:
Hey, I wonder if this issue is still available for GSoC 2018 xD
If so, it'd be great to have a hint where to start xD

bubnikv, 2018-03-21:
We at Prusa3D plan to give it as a first assignment to a guy joining us
this week.

FatemaKhalid, 2018-03-21:
Thank you for fast response :smile:

Samir55, 2018-06-17:
@bubnikv Has this task been implemented in Prusa3D, so we can port this to Slic3r ?

bubnikv, 2018-06-17:
We are still working on that.

tamasmeszaros, 2018-11-02:
Hello,
an implementation of the nesting has been developed in the form a C++ library. It is inspired by SVGNest but not based on it in any way. Slic3r Prusa Edition 1.4.1 has the first working version integrated.

At this stage it works only for convex polygons which was sufficient for the first release. A proper NFP implementation is still a work in progress. A have tried to integrate kallaballa's libnfporb library but it crashed with some examples.

At its current stage the nesting library lives on github: https://github.com/tamasmeszaros/libnest2d 
Any improvements would be welcome, especially a robust implementation of the no fit polygon for non-convex shapes with holes.

jjqcat, 2018-11-25:
@tamasmeszaros,
 I will try to use  libnfporb  in  libnest2d , but can't pack anything, should you give me some example code?thanks.

tamasmeszaros, 2019-03-04:
@jjqcat: Sorry for the late reply. If you have any trouble with libnest2d, please let me know I will try my best to help you. Also, if you make any progress with the libnfporb integration, please feel free to share your results on the libnest2d github page. It would be a valuable info in any case.

kallaballa, 2021-09-11:
I'd like to bring to your attention that I've finally put the necessary work into libnfporb to make it robust. There are still issues remaining: https://github.com/kallaballa/libnfporb/issues but they are rare and not critical. Also floating-point and arbitrary precision both yield the same result in the tests though that parity might break with a mix of very small and very large geometries.